ratgdo/homekit-ratgdo32 #1: Add required release workflow

dkerr64, 2024-11-30:
Need to update the workflow scripts to publish ratgdo32 version.

dkerr64, 2024-12-01:
@PaulWieland and @jgstroud we are in need of assistance.  @donavanbecker ran into a problem that I had also run into during development.  My workaround does not appear to work when we try and build on GitHub.  We get this build error...
```
Compiling .pio/build/esp32dev/lib81d/WiFi/AP.cpp.o
In file included from /home/david/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiSTA.h:29,
                 from /home/david/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFi.h:33,
                 from /home/david/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/AP.cpp:7:
/home/david/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiGeneric.h:39:10: fatal error: Network.h: No such file or directory

*****************************************************************
* Looking for Network.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:Network.h"
* Web  > https://registry.platformio.org/search?q=header:Network.h
*
*****************************************************************

   39 | #include "Network.h"
      |          ^~~~~~~~~~~
compilation terminated.
*** [.pio/build/esp32dev/lib81d/WiFi/AP.cpp.o] Error 1
```
Okay you say, that is easy.   Just add `Network` to the lib_deps in platformio.ini.  Except that doesn't work... I get a warning error when the build system looks for dependencies, see warning about cannot find the package.
```
PACKAGES: 
 - framework-arduinoespressif32 @ 3.0.7 
 - framework-arduinoespressif32-libs @ 5.1.0+sha.9439479fe5 
 - tool-esptoolpy @ 4.8.1 
 - tool-mklittlefs @ 3.2.0 
 - tool-riscv32-esp-elf-gdb @ 12.1.0+20221002 
 - tool-xtensa-esp-elf-gdb @ 12.1.0+20221002 
 - toolchain-xtensa-esp32 @ 12.2.0+20230208
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ deep+, Compatibility ~ off
Library Manager: Installing Network
Warning! Could not find the package with 'Network' requirements for your system 'linux_x86_64'
Found 44 compatible libraries
Scanning dependencies...
Dependency Graph
|-- EspSoftwareSerial @ 8.1.0+sha.57f1aa7
|-- HomeSpan @ 2.0.0+sha.30c3547
|-- WiFi @ 3.0.7
|-- DNSServer @ 3.0.7
|-- FS @ 3.0.7
|-- SPI @ 3.0.7
|-- Ticker @ 3.0.7
|-- ArduinoOTA @ 3.0.7
|-- Networking @ 3.0.7
|-- Update @ 3.0.7
|-- ratgdo
|-- Secplus @ 1.0.0
|-- WebServer @ 3.0.7
|-- HTTPClient @ 3.0.7
|-- NetworkClientSecure @ 3.0.7
Building in release mode
```
Changing from `Network` to `Networking` removes that warning, but does not solve the unable to find `Network.h` error.

My workaround was to add an explicit include path to the build in platformio.ini...
```
-I../../.platformio/packages/framework-arduinoespressif32/libraries/Network/src
```
Except this workaround is not working when we do the release build on GitHub.

Any ideas?

PaulWieland, 2024-12-01:
Worth trying?
https://github.com/espressif/arduino-esp32/issues/9782#issuecomment-2150947910

Am away until the end of next week so I cant easily test right now.

dkerr64, 2024-12-01:
Okay, wow yes.  My searching failed to turn up that discussion.  Adding that build flag appears to work.

Thanks.

PaulWieland, 2024-12-01:
I added it to platform.ini from my phone without realizing that you were simultaneously doing the same. I reverted mine. 
I think we can deep six all of the discord stuff in the workflow file...

closed 2024-12-05 by PaulWieland