ratgdo/homekit-ratgdo32 #2: Handle crashdump

dkerr64, 2024-11-30:
The ESP32 does not have a crash callback like ESP8266 but it does have crash dump... a 64KB block of flash into which process data is dumped at time of crash.

The firmware currently detects that a crash dump exists, and allows for erasing it.  But does not yet provide any way to view or download the contents of the crash dump.

mullender, 2024-12-31:
I am noticing the esp32 crashes a few times per day, I suspect it happens when the push button detects motion.

Is there a way I could capture the crash logs to facilitate debugging?

dkerr64, 2025-01-01:
ok, so it is concerning that you are seeing this... mine has been very stable.  As you probably can tell from this issue, the ESP32 is saving the crash information (and the web page should note this at the very bottom).  But we have not implemented any way to retrieve the saved crash data.

If you are seeing frequent crashes, then that is motivation for us to work on that.

In the meantime, if you can figure out a way to reliably recreate the crash, I can try it here.

Oh... and are you using Sec+ 2.0, 1.0 or dry contact?

Thanks.

mullender, 2025-01-01:
I’m using sec+ 2.0

I will keep trying to figure a consistent repro. But right now it’s been running 7+hours without issue. So that’s promising.

dkerr64, 2025-01-03:
Support for retrieving crash log data is added in PR #26

dkerr64, 2025-01-05:
Fixed in v3.0.5

closed 2025-01-05 by dkerr64

mullender, 2025-01-07:
Found a first crash log:
```
Messages logged immediately before last system crash...
 Time of crash: 05-Jan-2025 22:05:48 PST
Crash reason: Interrupt wdt timeout on CPU0
Firmware version: 3.0.5

00001193 000000E09B1D0126 0260C181
>>> [  13508.527] ratgdo-packet: PACKET(0x1D0126 @ 0x1193) Status - Status: [DoorState Open, Parity 0xC, Obs 1, Lock 0, Light 1]
>>> [  13508.547] ratgdo-comms: tgt 0 curr 0
>>> [  13512.745] ratgdo-reader: reader completed packet
>>> [  13512.745] ratgdo-packet: DECODED  00000FD5 000000622E9CA914 01018280
>>> [  13512.746] ratgdo-packet: PACKET(0x9CA914 @ 0xFD5) DoorAction - DoorAction: [DoorAction Toggle, Pressed 1, Id 01]
>>> [  13512.756] ratgdo-comms: Door Action
>>> [  13512.767] ratgdo-comms: Push count reset
>>> [  13512.767] ratgdo-comms: Push count 0
>>> [  13512.781] ratgdo-homekit: Motion detected
>>> [  13512.888] ratgdo-reader: reader completed packet
>>> [  13512.888] ratgdo-packet: DECODED  00001194 000000E09B1D0126 000931A1
>>> [  13512.889] ratgdo-packet: PACKET(0x1D0126 @ 0x1194) Pair3Resp - NoData: [Zero: 0x00090100, Parity: 0x3]
>>> [  13512.899] ratgdo-comms: Support for Pair3Resp packet unimplemented. Ignoring.
>>> [  13513.020] ratgdo-reader: reader completed packet
>>> [  13513.020] ratgdo-packet: DECODED  00000FD5 000000622E9CA914 01009280
>>> [  13513.021] ratgdo-packet: PACKET(0x9CA914 @ 0xFD5) DoorAction - DoorAction: [DoorAction Toggle, Pressed 0, Id 01]
>>> [  13513.031] ratgdo-comms: Door Action
>>> [  13513.412] ratgdo-reader: reader completed packet
>>> [  13513.413] ratgdo-packet: DECODED  00000FD6 000000622E9CA914 0000F085
>>> [  13513.413] ratgdo-packet: PACKET(0x9CA914 @ 0xFD6) Motion - NoData: [Zero: 0x00000000, Parity: 0xF]
>>> [  13513.423] ratgdo-comms: Motion Detected
>>> [  13513.455] ratgdo-packet: ENCODING 00000AE8 0000000000FBE539 00000080
>>> [  13513.587] ratgdo-reader: reader completed packet
>>> [  13513.587] ratgdo-packet: DECODED  00000AE8 0000000000FBE539 00008080
>>> [  13513.588] ratgdo-packet: PACKET(0xFBE539 @ 0xAE8) GetStatus - NoData: [Zero: 0x00000000, Parity: 0x8]
>>> [  13513.598] ratgdo-comms: Support for GetStatus packet unimplemented. Ignoring.
>>> [  13513.719] ratgdo-reader: reader completed packet
!!! [  13513.719] ratgdo-packet: Failed to decode packet
>>> [  13513.719] ratgdo-packet: DECODED  00000000 0000000000000000 00000000
>>> [  13513.730] ratgdo-packet: PACKET(0x0 @ 0x0) UNKNOWN - Unknown: [000]
>>> [  13513.740] ratgdo-comms: Support for UNKNOWN packet unimplemented. Ignoring.
>>> [  13513.852] ratgdo-reader: reader completed packet
>>> [  13513.852] ratgdo-packet: DECODED  00001195 000000E09B1D0126 02608581
>>> [  13513.853] ratgdo-packet: PACKET(0x1D0126 @ 0x1195) Status - Status: [DoorState Closing, Parity 0x8, Obs 1, Lock 0, Light 1]
>>> [  13513.873] ratgdo-comms: tgt 1 curr 3
>>> [  13513.874] ratgdo-vehicle: Vehicle status: Arriving
>>> [  13513.874] ratgdo-http: Current Door State changing from 0 to 3
>>> [  13513.885] ratgdo-homekit: Garage door set CurrentDoorState: 3
>>> [  13513.886] ratgdo-homekit: Arriving detected
>>> [  13513.902] ratgdo-homekit: Garage door set TargetDoorState: 1
>>> [  13513.910] ratgdo-vehicle: Vehicle status: Parked
>>> [  13513.911] ratgdo-homekit: Arriving reset

Crash in task: pollTask, at address: 0x400847B1
Decode backtrace with this Linux command:

addr2line -p -f -C -e homekit-ratgdo32-v3.0.5.elf \
 -a 0x400847B1 0x400848E5 0x401054C3 0x400DD669 0x400E405A 0x400DDC91 \
    0x400DCFD8 0x400F7E16 0x400DA8F2 0x40090512 

Make sure that the ELF file matches the binary that crashed.
```

dkerr64, 2025-01-07:
I saw this myself a few days ago.  I have to think about how to fix it.

Cause is Watchdog Timeout (wdt) on CPU0.  There are a couple of tasks in ESP32 that monitor CPU0 and CPU1 to make sure they do not get into a hung state (or, say, an infinite loop).  Something is taking too long.  Or I need to find out how to make the watch dog timer more tolerant (allow more time before thinking there is a problem).

We should really open this as a fresh issue.

mullender, 2025-01-07:
Opened #30 to track this specific crash