ratgdo/homekit-ratgdo32 #4: Add support for the beeper

dkerr64, 2024-11-30:
Add support for the beeper... with option to activate it during time-to-close delay.

Should the beeper be reported as a HomeKit service?  If so what?

PaulWieland, 2024-11-30:
I don't think the beeper needs to be a HomeKit service.

dkerr64, 2024-12-07:
Feature will be added in PR #11

Tone is aways active, no option to disable (we have no option to disable light flashing, so why add it for the beeper).

I looked at RTTTL, there is a library for that (I was testing with [this one](https://github.com/erik-toth/ESP32-RTTTL)) but ultimately decided to go with simple tones using Arduino built-in tone generator.

- A tone on boot, with slightly higher pitch tone when initialization completes.
- A tone during time-to-close with slightly higher pitch when TTC ends and door closes.

closed 2024-12-12 by dkerr64