ratgdo/homekit-ratgdo32 #12: a really stupid question

marcosharkswfl, 2024-12-09:
Mac user here. I accidentally removed my Ratgdo from my HomeKit screen and I can't seem to add it back. How does one do a reset? I try to do a re install of the firmware and I am asked to hold down a boot button? 
[image]

What am I missing here? is it something so basic? HELP!

(thanks!!!!!)

dkerr64, 2024-12-09:
First make sure you are using the correct firmware for the hardware you have.  Do you have the new "ESP32" version of ratgdo (also known as disco) or the older ESP8266 version?  It is important to use the correct firmware installer for each version. 

You have opened this question on the newer ESP32 / Disco firmware page.  This new firmware has a "factory reset" button on the setting page.

Both the older and newer devices have an "Un-pair HomeKit" or a "Reset HomeKit" button on the main page.  That would be an alternative to a factory reset.

Both the above assume that the ratgdo firmware is booting up and you can reach it through a browser.  If that is not the case then reinstall the correct firmware, from the correct installer, for the hardware version you are using.  The message you included suggests that you are at the wrong installer for your h/w.  There is no "boot" button on any version of the h/w.

Thanks

dkerr64, 2024-12-11:
@marcosharkswfl have you solved your problem?

dkerr64, 2024-12-12:
Closing as I don't believe there is a bug.

closed 2024-12-12 by dkerr64