ratgdo/homekit-ratgdo32 #13: Build our own ESP32 framework

dkerr64, 2024-12-12:
The size of our binary is creeping up.  With addition of vehicle occupancy the binary is now over 90% of available space.

So I investigated whether we can tell Arduino / ESP not to compile pieces we don't use... in other words use menuconfig to configure what components we want.  It turns out that Arduino framework defaults to a pre-compiled default ESP32 framework.  To change that we can add a `sdkconfig.defaults` file and make minor tweaks to `platform.ini`.  There is an example [here](https://github.com/Jason2866/platform-espressif32/tree/Arduino/IDF5/examples/espidf-arduino-blink).

I have tried this and modified the example sdkconfig to this...
```
CONFIG_AUTOSTART_ARDUINO=y
# CONFIG_WS2812_LED_ENABLE is not set
CONFIG_FREERTOS_HZ=1000
CONFIG_MBEDTLS_PSK_MODES=y
CONFIG_MBEDTLS_KEY_EXCHANGE_PSK=y
CONFIG_ESP_COREDUMP_ENABLE_TO_FLASH=y
CONFIG_ESP_HTTPS_SERVER_ENABLE=n
CONFIG_ESP_HTTP_CLIENT_ENABLE_HTTPS=n
CONFIG_ESP_INSIGHTS_TRANSPORT_HTTPS=n
CONFIG_ESPTOOLPY_FLASHSIZE_4MB=y
CONFIG_LOG_COLORS=n
CONFIG_LOG_DEFAULT_LEVEL_NONE=y
```
and platformio.ini to include...
```
board_build.embed_txtfiles =
    managed_components/espressif__esp_insights/server_certs/https_server.crt
    managed_components/espressif__esp_rainmaker/server_certs/rmaker_mqtt_server.crt
    managed_components/espressif__esp_rainmaker/server_certs/rmaker_claim_service_server.crt
    managed_components/espressif__esp_rainmaker/server_certs/rmaker_ota_server.crt
```
and
```
framework = arduino, espidf
```

The downside to this is that builds take significantly longer, even incremental builds.

PaulWieland, 2024-12-12:
I wonder what makes this firmware so bloated... the [ESP Home disco firmware](https://ratgdo.github.io/esphome-ratgdo/v32disco-esp32/v32disco-esp32.factory.bin) is only 1.1MB.

dkerr64, 2024-12-14:
I have added the necessary changes to PR #16 but it is not enabled because it adds considerable time to the builds.

In addition to the space savings, we can also enable CONFIG_FREERTOS_USE_TRACE_FACILITY which gives us the ability to monitor FreeRTOS task memory usage.  But that is niche so leave disabled.

closed 2025-09-13 by dkerr64