ratgdo/homekit-ratgdo32 #14: Browser-based flash tool not working.

dkerr64, 2024-12-12:
@PaulWieland @donavanbecker The web browser based flash tool is not working.  I have tried to install 3.0.1 on both the ESP32-based boards I have and both fail.  The upload appears to work successfully but it does not boot.  Below is the log which just repeats continuously...

The worst part is that it seems to blast over the entire flash... I did not select "erase" but when I recovered by uploading from the VScode platformio, all settings in NVRAM had gone so I had to initialize from scratch... WiFi and re-pair to HomeKit.

Can either of you assist?  Thanks.

```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
invalid header: 0x756c2520
invalid header: 0x756c2520
invalid header: 0x756c2520
invalid header: 0x756c2520
invalid header: 0x756c2520
invalid header: 0x756c2520
invalid header: 0x756c2520
ets Jul 29 2019 12:21:46
```

PaulWieland, 2024-12-12:
I just downloaded [`homekit-ratgdo32-v3.0.1.bin`](https://ratgdo.github.io/homekit-ratgdo32/firmware/homekit-ratgdo32-v3.0.1.bin) and then flashed it using `esp_tool.py` like this: `esptool.py write_flash 0 ~/Downloads/homekit-ratgdo32-v3.0.1.bin`

When I connect to the console, I'm getting the same error.

dkerr64, 2024-12-12:
Humm... odd.  I'll investigate more from my side.

dkerr64, 2024-12-12:
I think I know what is going on.  The firmware binary represents the executable code, not an image of the flash.  The flash is divided into multiple partitions defined in `partitions.csv`...

```
# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x5000,
otadata,  data, ota,     0xe000,  0x2000,
app0,     app,  ota_0,   0x10000, 0x1F0000,
app1,     app,  ota_1,   0x200000,0x1F0000,
coredump, data, coredump,0x3F0000,0x10000
```

Writing to address 0 blasts over reserved sections of the flash, and the NVS storage area.

I believe that what we must do is write the firmware to the app0/ota_0 partition (so offset 0x10000) and erase the otadata/ota partition (at offset 0x0e000).  The erase is needed to reset the boot loader OTA data, as this data determines whether to execute the firmware in the first or second OTA partition... erasing it should force it to reset and use the first partition.

I will test this later today.

PaulWieland, 2024-12-12:
FWIW, the [manifest](https://ratgdo.github.io/esphome-ratgdo/v32disco-manifest.json) for the esp home firmware writes to address 0x00. But it also has a separate binary for OTA, not quite sure why yet.

dkerr64, 2024-12-12:
Here is what the platformIO runs to upload flash. 

```
"/home/david/.platformio/penv/bin/python" "/home/david/.platformio/packages/tool-esptoolpy@src-0fed74e9a0661ea9c83289dd49725739/esptool.py" \
--chip esp32 --port "/dev/ttyACM0" --baud 921600 --before default_reset --after hard_reset \
write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect \
0x1000 /home/david/github/homekit-ratgdo32/.pio/build/ratgdo_esp32dev/bootloader.bin \
0x8000 /home/david/github/homekit-ratgdo32/.pio/build/ratgdo_esp32dev/partitions.bin \
0xe000 /home/david/.platformio/packages/framework-arduinoespressif32/tools/partitions/boot_app0.bin \
0x10000 .pio/build/ratgdo_esp32dev/firmware.bin
```

I split it onto multiple lines here for readability.

But bottom line is we need to do the same in our flash installer tool.  The boot loader, partitions and firmware are all built as part of ratgdo build.  The OTA section is obtained from the framework.  The hardcoded offsets here will have been read from `partitions.csv`, so if that ever changes then we will need to change the flash installer.  I don't think we ever want to do that! which is why I set the partitions to have maximum possible space available for executable binary.

dkerr64, 2024-12-12:
> FWIW, the [manifest](https://ratgdo.github.io/esphome-ratgdo/v32disco-manifest.json) for the esp home firmware writes to address 0x00. But it also has a separate binary for OTA, not quite sure why yet.

I think one version probably includes the boot loader, partitions and otadata that will go in from address 0x00, with the executable firmware appended.  And the OTA version is just the executable.  I assume that the esphome partition table does not have NVS squeezed into the first 64KB like I did, else it will get erased every time flash is uploaded.

dkerr64, 2024-12-13:
With @donavanbecker help we have web flash tool working.  But I need to make a change to the browser side javascript for OTA updates before releasing (as we change file name of the binary).

PaulWieland, 2024-12-13:
What was the culprit?

donavanbecker, 2024-12-13:
> What was the culprit?

the offsets and getting all the needed files represented In the manifest.json

dkerr64, 2024-12-13:
This should all be working now.  Have published a release with the needed fixes and updated the readme and flash installer with instructions on how to setup WiFi network and password.  OTA update is working too.

We can now allow public testing.

closed 2024-12-13 by dkerr64

risk235, 2024-12-14:
Just wanted to chime in and thank y’all for your hard work. I just finished installing mine and I’m able to control my garage in the Home app.