ratgdo/homekit-ratgdo32 #23: Add a light switch for the parking assist laser

dkerr64, 2024-12-20:
Currently parking assist laser is coded to come on for 60 seconds when arriving vehicle arrives.

However, I am finding that the vehicle presence (and therefor arrival/departure) sensing is very inconsistent.

So, should we add a light switch accessory to HomeKit (and a button on the device web page) to allow manual control of the laser... and of course, inclusion in HomeKit automations?

PaulWieland, 2024-12-20:
I have a manual switch in esphome which is helpful during initial setup when you are aiming the sensors. I have to setup another board to do a long term test with the HK firmware, but I can definitely say that the esp home firmware has been extremely reliable.

dkerr64, 2024-12-20:
Added in PR #24

dkerr64, 2024-12-29:
PR #24 is merged, ability to enable/disable added to PR #26

closed 2024-12-29 by dkerr64