ratgdo/homekit-ratgdo32 #28: Door close status feedback to HomeKit Automation

mtmcooper, 2024-12-29:
Homekit automation process does not appear to detect the current state of the door.  I have an automation that closes the door.  If the door is already closed, HomeKit indicates that the door is "closing" and continues to show this status even though the door is already closed.

donavanbecker, 2024-12-29:
This must be an iOS bug, as I am experiencing the same and wasn't before hand.

dkerr64, 2025-01-01:
Thanks for reporting this,  I will try and reproduce.  Can you tell me if you are Sec+ 2.0 or 1.0 please.  And I assume your automation is sending a "door close" even though the door is already closed, and that is when you see the problem (ie, it works as expected if the door is open).

Thanks.

mtmcooper, 2025-01-01:
Hi David,I have Sec+2.0 Chamberlin openers.Yes, the automation was sending the “Door Close” signal even though the door was already closed.  This was done by adding the garage door to an automation that runs at a specific time in the evening.I should also share that I initially had a delay time set for the door close.  I have since set the time to 0.  When I had the time delay the controller would forget the WiFi setup when the automation was initiated.  Turning off the delay seemed to fix that issue. Mark Cooper

dkerr64, 2025-01-01:
I am able to reproduce and I see where the bug is.  Will be fixed when PR #26 is merged and released.

Thanks for reporting this.

dkerr64, 2025-01-05:
Fixed in v3.0.5

closed 2025-01-05 by dkerr64