ratgdo/homekit-ratgdo32 #33: Enhancement request: Signal "beep" at the same time as the light when Door Close Delay

jakepez, 2025-01-14:
Enhancement request: 
Signal "beep" at the same time as the light flash when Door Close Delay set
Minimum time for Door Close Delay - 5 seconds (or not in the interface that 5 seconds is the US Federal requirement.

This will allow conformance with [16 CFR § 1211.14](https://www.law.cornell.edu/cfr/text/16/1211.14)

dkerr64, 2025-01-14:
The beep is supposed to be working.  However it may not be very loud as the buzzer on the ratgdo32 device is not very loud.

Time-to-close delay is configurable, from zero to 60 seconds. I think what you are saying is that the minimum should be 5 seconds.  This is of course possible but may trigger some dissent from some users who dislike the delay.  But as you point out, there is legal guidelines we should follow.

Thanks.

jakepez, 2025-01-14:
Agree on the dissent - hence, maybe a warning in the interface (or force to override the minimum)?

For the “beep” my opener has it’s own speaker/alarm/beep - which is triggered if I use the myQ app remote or my existing myQ Home Bridge (which I’m trying to get rid of as it’s super flaky and needs to be power cycled every few days). I was thinking since the myQ app can do this - maybe it could be triggered just like the lights are.

jakepez, 2025-01-14:
I’m wondering - and getting lost as I try to understand GDO communications ;) - if the folks over at Konnected are using a Time to Close command directly to the Garage Door Opener (rather than their own timer?) - and if that’s how they are supporting the US Federal requirements?
[https://github.com/konnected-io/gdolib/blob/6ab913b551d6add622398b83fa759f67616a69bd/gdo_priv.h#L59](https://github.com/konnected-io/gdolib/blob/6ab913b551d6add622398b83fa759f67616a69bd/gdo_priv.h#L59)

dkerr64, 2025-01-14:
I think maybe I default to 5 seconds and if a user selects a shorter time then issue a popup warning.

There has been quite a bit of discussion on time-to-close over on the ratgdo GitHub, both [ESPhome](https://github.com/ratgdo/esphome-ratgdo/issues/42) and [HomeKit](https://github.com/ratgdo/homekit-ratgdo/issues/230) and the teams could not figure out how to get the built-in command to work.  But I have not personally tried, so maybe the pointer you give to Konnected will help.

In the meantime we are manually triggering flashing the light, and as there is no API control for the GDO's built-in beeper, using the beeper added to the ratgdo32 board (the original ESP8266-based board does not have it).

jakepez, 2025-01-14:
I’ll order the Konnected BlaQ device and see how it works. I’ll let you know if it triggers the GDO alert with the built in command or not.

dkerr64, 2025-01-18:
Fixed in v3.0.6

closed 2025-01-18 by dkerr64