ratgdo/homekit-ratgdo32 #35: V3.0.6 causes Ratgdo32 to freeze up with LiftMaster 8500 (Sec+)

Slabonski, 2025-02-02:
Both McNasty528 and I are having the same problem (see Reddit link below). V3.0.6 does track the status of the garage door and light of our LIftMaster 8500 (Sec+) opener in both the web interface and HomeKit. As soon as we attempt to use either the web interface or the Home app to open/close the door or turn on/off the light the Ratgdo32 will execute that one command and then freeze up. The web interface will no longer load and the home app just shows either the prior state (if using web interface) or closing/opening if using Home app. The only way to get it back is to unplug it and plug it back in again. All normal door buttons and obstruction sensors work fine. McNasty528 was able to revert his back to his prior V3.0.1 to get it working again. Not seeing any crash log. Here is the Reddit link https://www.reddit.com/r/ratgdo/comments/1i6v9ds/need_help_ratgdo32/

dkerr64, 2025-02-02:
Thanks for opening the issue.

Are you able to test this while capturing logs from the serial port, or setting up a syslog server?

Slabonski, 2025-02-02:
Yes I should be able to setup a syslog server. I saw the syslog settings in the web interface. I will try to capture the logs while it is freezing up.

Slabonski, 2025-02-02:
Ok I think I was able to successfully capture the syslog output on my Mac using tcpdump piped to strings. These are some test captures with different scenarios described at the top of each file. Let me know if you want me to do anything specially.

[Door open at start using HK to try to close.txt](https://github.com/user-attachments/files/18633725/Door.open.at.start.using.HK.to.try.to.close.txt)
[Door open light on attempt control home app.txt](https://github.com/user-attachments/files/18633728/Door.open.light.on.attempt.control.home.app.txt)
[Door open light on turn light off web interface.txt](https://github.com/user-attachments/files/18633727/Door.open.light.on.turn.light.off.web.interface.txt)
[Repower then HK door open but did not open door first.txt](https://github.com/user-attachments/files/18633729/Repower.then.HK.door.open.but.did.not.open.door.first.txt)
[Repower then Homekit door open.txt](https://github.com/user-attachments/files/18633726/Repower.then.Homekit.door.open.txt)
[Repower then open door using web interface.txt](https://github.com/user-attachments/files/18633724/Repower.then.open.door.using.web.interface.txt)

Slabonski, 2025-02-02:
I should note that my light is a Liftmaster remote light switch 823LM.

Slabonski, 2025-02-02:
So it does not look like it completely freezes up. If I repower the RATDGO32 and then open the door with either the Home app or the web interface the garage door will usually open. After it is opened, the web interface no longer responds but I do get some syslog responses from the home app. The status in the Home app never changes from opening to opened. If I hit it again it just says closed. I can repeat this over and over and I will get syslog responses but the door stays open. Same with the light. As you can see from the other logs it responds similarly (web interface freezes some responses from Home app) when the door starts open but the door never closes.

[Door closed light off use web interface to open then try Home app.txt](https://github.com/user-attachments/files/18634046/Door.closed.light.off.use.web.interface.to.open.then.try.Home.app.txt)
[Door closed light off use Home app to open then try to close.txt](https://github.com/user-attachments/files/18634045/Door.closed.light.off.use.Home.app.to.open.then.try.to.close.txt)

dkerr64, 2025-02-03:
Thanks for the logs. In do see quite a few that are of concern which gives clues for me to look at. 

Could you reduce the time-to-close delay to zero please… just to remove one variable, some of the errors are coming up in that code path.

Slabonski, 2025-02-03:
I can do that but I will not be available to do more testing until next weekend.

MichaelCWarren, 2025-02-07:
I'm getting the same issue.  Any action causes the door to freeze up for about 30 seconds.  Even actions using the wall button (light on/off, open/close) causes the issue.

I'm more than willing to try a beta.  

I'll give the TTC delay a try.

MichaelCWarren, 2025-02-07:
Update!

I just seet the TTC delay to 0, and it worked perfectly.  

I'll use it for a couple of days and let you know.

dkerr64, 2025-02-07:
Thanks for the report.  I do know what is causing it, and I need to think how best to fix.

When TTC is active, the ratgdo sends a series of commands to the door to turn the light on and off... spaced out half a second.  These commands are added to a queue.  It appears that the door is not accepting the commands as fast as I am sending them, so each request is removed off the queue at a slower pace than they are added.  Logs appended above show that the queue fills up and errors start to log... and I have not implemented a recovery.

The work around for now is to set TTC to zero.  But I will look into a proper fix.

Thanks.

dkerr64, 2025-02-08:
I believe the problem is the special handling we put in for Sec+1.0 garage doors with a digital wall panel.  That appears to not be  working.  I have changed the code such that we handle sending of commands the same whether you have a digital wall panel or not.  I would like you to test this.  It will be in v3.0.7 that I am marking as "pre-release" so you will need to check that box on the update window.

Thanks.

cramlr, 2025-02-25:
@dkerr64 I too have a lift master 8500 with myQ wall control.  I am using firmware 3.0.7 but still have the same issue that the ratgdo disco won't respond to the web ui or HomeKit.  

I set the TTC to 0 and now it is working OK.  So it doesn't appear the fix worked.

cramlr, 2025-02-25:
Since I have Home Assistant, should I be using a different firmware instead of the HomeKit one?

Happy to help troubleshoot

dkerr64, 2025-03-08:
Please try release 3.1.0 and advise if this is working.

Slabonski, 2025-03-10:
Just tried 3.1.2 and it works now using HomeKit as long as the close delay is zero. If I use the default of 5 secs it will open the door but not close it. That is fine for me since I don't want a delay and have a camera to verify no one is in the way.

dkerr64, 2025-03-11:
Can you provide the logs please.

Slabonski, 2025-03-11:
> Can you provide the logs please.

I can but will not have a chance for at least a week.

dkerr64, 2025-03-11:
Also, is this Sec+ 1.0 or 2.0.  Thx.

dkerr64, 2025-09-13:
Please test with pre-release v3.3.1 (select pre-release box on update dialog).  If you are unable to update from GitHub please download firmware from [here](https://github.com/ratgdo/homekit-ratgdo32/releases/download/v3.3.1/homekit-ratgdo32-v3.3.1.firmware.bin) and install from computer.

Closing this issue, please open a new issue if problems found.

closed 2025-09-13 by dkerr64