ratgdo/homekit-ratgdo32 #39: Vehicle Distance Actual

Idowza, 2025-02-19:
I recently installed the Disco and I'm really enjoying it. Everything is working perfectly except for the "Vehicle Distance Actual" sensor. It's fluctuating between 750 and 780mm, but then occasionally jumps to 4500mm for a second or two. This jump also causes the "Vehicle Detected" status to switch from ON to OFF. Any ideas what might be causing this or how to fix it?

dkerr64, 2025-02-19:
Check to see that the distance sensor does not have a tiny yellow sticker over the two (transmit and receive) holes.  This is to protect the sensors from dust, etc. during manufacturing and if it was not removed before shipping could be interfering with the sensors.

The sensor looks something like this...

[image]

Idowza, 2025-02-19:
This is what mine looks like.  I don't see any stickers. 

[image]

Idowza, 2025-02-19:
When I try to shorten the distance by placing an object, the reading remains stable up to around 600mm.  Beyond that, it abruptly jumps back to 4500mm.

dkerr64, 2025-02-19:
I'm not seeing anything obstructing the sensors either.  Your best bet is to ask @PaulWieland if he has any ideas.

PaulWieland, 2025-02-19:
Which firmware are you using? you posted the same issue to both HK and ESP Home repos.

closed 2025-02-19 by PaulWieland