ratgdo/homekit-ratgdo32 #40: Feature Request

donavanbecker, 2025-02-19:
Add an option to disable Arriving, Departing, and Vehicle Motion Sensors.

dkerr64, 2025-02-19:
So, you are asking for an option, similar to the checkbox to enable a HomeKit light switch for the laser, but to control creating the motion sensor accessories in HomeKit for the vehicle status?

Would be fairly easy to do... but is there a good reason to do it?  Feels harmless to always have the HomeKit accessories created?

donavanbecker, 2025-02-19:
I just got a wall mounted garage door opener and I have the ratgdo32 disco, currently it is hidden behind the cover of the opener so the vehicle present sensor is showing as motion detected all the time.

dkerr64, 2025-02-19:
Ok, so yes in that case it would be useful to disable the HomeKit accessories.  I think it is fairly easy to implement.

donavanbecker, 2025-02-20:
Just tested or, working awesome

dkerr64, 2025-02-20:
Thanks.  Though every time I add a user interface setting it touches way more source files than you might think !!!

Hoping to push a pre-release for testing soon.

donavanbecker, 2025-02-28:
This can be closed as it is confirmed working and in the next release PR.

closed 2025-02-28 by donavanbecker