ratgdo/homekit-ratgdo32 #41: Homekit Dry Contact version - only appears to open or close the door by using the close door command or HomeKit slider position

Akadan1999, 2025-02-22:
Using HomeKit firmware on dry contact garage door opener.  
Homekit only seems to activate the door by using the close command.  This may be correct since this simulates a simple 
toogle switch.  Homekit visual status of door position is also somewhat random since there may be no feedback from the unit to indicate status.
For example the door is physically closed and the visual indicates closing.  Oral command to open door does nothing physically or visually in HomeKit.  Oral command to close the door activates opening the door physically, but indicates closing in the HomeKit visual.  Touching the HomeKit visual slider and moving to Open just changes the test to Open, nothing physical changes.  Sliding to the close position in the door is physically open will close the door and text on the slider indicates Closing but never Closed even if the door is physically closed.

The obstruction safety functions work physically as expected in this mode but Homekit does not reflect Obstruction detected.

dkerr64, 2025-02-24:
There are several problems with dry contact... depending on the version your are running. You might want to try the 3.0.7 pre-release if you are not on that version.

Also tagging @donavanbecker as he is our resident expert on dry contact (or at least, has greatly helped get it where it is today).

donavanbecker, 2025-02-25:
Obstructions aren't working for me either.

@Akadan1999 I am guessing you are talking about the ratgdo UI? I have had really good success with the Home App showing the right status; but in ratgdo UI, all I see is Unknown for closed door status.

PaulWieland, 2025-02-27:
I can also confirm that obstruction detection isn't working.

Regarding the door state, upon initial boot up, the door state is unknown. The door state should only be unknown if both limit switches are open. Otherwise the door state is equal to whichever limit switch is closed.

dkerr64, 2025-02-27:
I'm working on a fix for next release.

dkerr64, 2025-03-08:
Please test with release 3.1.0

dkerr64, 2025-04-01:
Please test with release 3.1.6.

dkerr64, 2025-04-27:
There were problems in obstruction sensing that are fixed (hopefully) in version 3.1.7

donavanbecker, 2025-04-27:
Working awesome for me at 3.1.6. I will test 3.1.7, obstruction worked most of the time already for me.