ratgdo/homekit-ratgdo32 #42: Feature Request: PWA Reload Button

donavanbecker, 2025-02-27:
Would be nice to have a reload button when webpage is in PWA mode so that you can force a refresh of the webpage, which you can't do even in a PWA.

dkerr64, 2025-02-27:
Just reload status, or reload all html, javascript, etc?

donavanbecker, 2025-02-27:
I would say at least all html. since sometimes it gets in a state and being able to click that reload would help recover.

dkerr64, 2025-02-28:
If I create an app icon with "Add to Home Screen" on iPhone or iPad... then when I am on that page, I can swipe down with my finger and the whole page will reload.  The web server code in ratgdo32 (that I wrote) is smart enough to send a "304 Not Modified" response to the client if the page has not changed (based on CRC code).

Is there another scenario to handle?

donavanbecker, 2025-02-28:
Didn't know you can pull down. I am good!

dkerr64, 2025-02-28:
> Didn't know you can pull down. I am good!

It takes some JavaScript to make it work... has been in our web page since very early days.