ratgdo/homekit-ratgdo32 #45: Feature Request: Occupancy Sensor Support for GDO

simplytoast1, 2025-03-08:
Description:
I would like to request the addition of occupancy sensor support to the GDO.

Use Case:
I often work in my garage and use supplemental lighting in addition to the GDO light. Currently, I have an automation that turns on the supplemental lighting when the GDO light is on. However, using a motion sensor as a trigger is not ideal due to its short activation window.

Proposed Solution:
Introduce a user-configurable timer that functions as an occupancy sensor. When motion is detected, the occupancy sensor is triggered, and the timer starts. If motion is detected again within the preset time, the timer resets, keeping the occupancy flag active in HomeKit. This would provide a more reliable way to determine occupancy without relying solely on short-duration motion sensor triggers.

dkerr64, 2025-03-08:
I think you could do this with an automation in Apple Home.... just use the motion sensor to turn on a light for a specified duration.

homekit-ratgdo32 already creates a motion accessory in HomeKit, I do not want to make it any more complex than it already is by adding multiple accessories into HomeKit for the same sensor.

egrim, 2025-03-12:
This [feature](https://github.com/hjdhjd/homebridge-ratgdo/blob/main/docs/FeatureOptions.md#opener-motion-feature-options) is available in @hjdhjd/homebridge-ratgdo and I found it very helpful to set up an automation to close the door if the door was open and no motion had been detected in the garage for a certain amount of time.  I plan to implement this for myself on a fork.  If there's any interest from others, I'll make it an option users can enable in the settings page and will make a pull request in case @dkerr64 might be willing to reconsider adding it to the main repo.

dkerr64, 2025-03-14:
If you do code this up and submit a PR, take a look at this [commit](https://github.com/ratgdo/homekit-ratgdo32/commit/10576557705d2ad2f39c848fc279d2cbe25f670f) for some hints on how to add a HomeKit accessory as an option that the user can select.

dkerr64, 2025-04-27:
This feature will be added in the next release

closed 2025-04-27 by dkerr64