ratgdo/homekit-ratgdo32 #47: ratgdo32 reporting incorrect door and light status

EricSihaoLin, 2025-03-09:
Hey there, recently got my hands on the Ratgdo and flashed it with v3.1.1 of the firmware via chrome. Just something I noticed when I’m setting it up according to the wiring diagram, the door status is not accurate as the analog buttons I use to control the door does not update the status of the ratgdo (neither does the light, but that’s a smaller issue)

I don’t see anything in the logs over WiFi that would indicate the status was changed by any of the physical buttons. I have gone ahead and attached some screenshot as reference in case if I am an idiot.

Since it is a purple learn button, I did change the protocol to security 1.0.

[image]
[image]

dkerr64, 2025-03-09:
Try something for me please... return the original wiring to connect directly to the door opener, then connect the ratgdo into the same connectors (ie, wire the ratgdo in parallel rather than in series).

dkerr64, 2025-03-09:
I am looking into the problem here and it does seem like I broke something with Sec+ 1.0.

dkerr64, 2025-03-09:
Thank you for reporting this bug.  It is fixed in release 3.1.2 that I just published (as pre-release, so be sure to select that option on firmware update).

closed 2025-03-09 by dkerr64

EricSihaoLin, 2025-03-09:
Thanks so much @dkerr64, after flashing 3.1.2, the problem is now fixed.