ratgdo/homekit-ratgdo32 #48: Occasionally ratgdo32 keeps on toggling status

EricSihaoLin, 2025-03-10:
I'm currently running v3.1.2 of the release.

Here's what I see in the system logs right before a reboot:

```
.638) ratgdo-http: Client 192.168.1.184 not listening, remove SSE subscription. Total subscribed: 1
I (6188.840) ratgdo-comms: GDO event: door: Closed, 100%, target:  -1%
I (6188.842) ratgdo-homekit: Garage door set CurrentDoorState: 1
I (6188.849) ratgdo-http: Current Door State changing from Open to Closed
I (6188.854) ratgdo-http: SSE send to client 192.168.1.184 on channel 1, data: { "lastDoorUpdateAt": 0, "garageDoorState": "Closed", "upTime": 6185798 }
W (6188.941) gdolib: Timeout in xTaskNotifyWait
W (6188.942) gdolib: Parity error 0xF0
W (6189.089) gdolib: Parity error 0xFF
W (6189.589) gdolib: Parity error 0xFF
I (6189.590) ratgdo-comms: GDO event: obstruction: Obstructed
I (6189.591) ratgdo-homekit: Garage door set ObstructionDetected: 1
I (6189.599) ratgdo-http: SSE send to client 192.168.1.184 on channel 1, data: { "garageObstructed": true, "upTime": 6186548 }
W (6189.942) gdolib: Timeout in xTaskNotifyWait
I (6190.840) ratgdo-comms: GDO event: obstruction: Clear
I (6190.842) ratgdo-homekit: Garage door set ObstructionDetected: 0
I (6190.847) ratgdo-http: SSE send to client 192.168.1.184 on channel 1, data: { "garageObstructed": false, "upTime": 6187796 }
W (6190.943) gdolib: Timeout in xTaskNotifyWait
I (6191.098) ratgdo-comms: GDO event: door: Open,   0%, target:  -1%
I (6191.099) ratgdo-http: Current Door State changing from Closed to Open
I (6191.109) ratgdo-homekit: Garage door set CurrentDoorState: 0
I (6191.115) ratgdo-http: SSE send to client 192.168.1.184 on channel 1, data: { "lastDoorUpdateAt": 0, "garageDoorState": "Open", "upTime": 6188048 }
I (6191.840) ratgdo-comms: GDO event: door: Closed, 100%, target:  -1%
I (6191.843) ratgdo-homekit: Garage door set CurrentDoorState: 1
I (6191.849) ratgdo-http: Current Door State changing from Open to Closed
I (6191.854) ratgdo-http: SSE send to client 192.168.1.184 on channel 1, data: { "lastDoorUpdateAt": 0, "garageDoorState": "Closed", "upTime": 6188798 }
W (6191.944) gdolib: Timeout in xTaskNotifyWait
W (6191.945) gdolib: Parity error 0xC0
W (6192.089) gdolib: Parity error 0xFF
I (6192.590) ratgdo-comms: GDO event: obstruction: Obstructed
I (6192.592) ratgdo-homekit: Garage door set ObstructionDetected: 1
I (6192.599) ratgdo-http: SSE send to client 192.168.1.184 on channel 1, data: { "garageObstructed": true, "upTime": 6189548 }
W (6192.945) gdolib: Timeout in xTaskNotifyWait
I (6192.946) ratgdo-comms: GDO event: obstruction: Clear
I (6192.947) ratgdo-homekit: Garage door set ObstructionDetected: 0
I (6192.947) ratgdo-http: SSE send to client 192.168.1.184 on channel 1, data: { "garageObstructed": false, "upTime": 6189896 }
W (6193.946) gdolib: Timeout in xTaskNotifyWait
W (6194.947) gdolib: Timeout in xTaskNotifyWait
I (6195.365) ratgdo-http: Client 192.168.1.184 requesting: /reboot (method: HTTP_POST)
I (6195.367) ratgdo-http: System boot time:    10-Mar-2025 16:47:52 EDT
I (6195.378) ratgdo-http: Reboot requested at: 10-Mar-2025 18:31:04 EDT
I (6195.884) ratgdo-comms: Save rolling code: 0
I (6195.885) ratgdo-logger: Save message log buffer
```

The door status constantly gets togged and I have no idea why. The actual physical door is not being opened or closed, only the ratgdo status being reported.

After a reboot, it seems okay. I'm still monitoring.
I'm using Sec 1.0 with a LiftMaster purple learn button.

dkerr64, 2025-03-11:
It might help to set the log level to debug. If ratgdo is emulating a wall panel then that is going to generate a lot more messages… but it might be the only way to identify the problem.

EricSihaoLin, 2025-03-11:
I'll report back with more logs when I'm awake for a longer period of time. (Most likely Wednesday this week)
For now I'll keep it unplugged during the night in case if it does something weird.

EricSihaoLin, 2025-03-12:
While I couldn't get the state to toggle yet, I did notice something weird around 1 hour in to a ratgdo power-on session.
Here's the logs I'm able to produce: https://gist.github.com/EricSihaoLin/bf76c9745ed094ab78f5922209d2e0f3
I'll keep trying for a log when it actually toggles the door state, but hopefully you're able to gleam something from this.

The parity errors starts to roll in around 3300 on the ratgdo timer, around line 11133 on the gist.

EricSihaoLin, 2025-03-15:
I have the ratgdo on a syslog with debug logging now.
Here's the latest crash, just today it crashed 9 times with similar messages:
https://gist.github.com/EricSihaoLin/483a46c6eb3ece7f733863d8cbc3739d
Let me know if getting these syslogs would be beneficial

EricSihaoLin, 2025-03-15:
I was just able to catch the ratgdo actually opening my garage door without being told to do so.
Here's the gist of the log:
https://gist.github.com/EricSihaoLin/d5d3bc6c332257868d034d37f30e1742
The door open was logged at March 14th, 2025 at 10:24 PM. The log was terminated at 10:28 PM as I physically disconnected the ratgdo device.

dkerr64, 2025-03-15:
Thanks for the log. Lots of error conditions in there, question is why it got itself into such a state. I’m not sure if the parity errors are the cause or a symptom.

Do you have anything else connected to the door other than the ratgdo and wall panel?

EricSihaoLin, 2025-03-15:
In terms of wired, just the analog wall buttons that controls this door which is going through the Ratgdo.
In terms of wireless, there’s a code panel on the outside that is mounted, and I currently still have a myQ monitoring this door until I’m fully confident in the rat.

EricSihaoLin, 2025-03-15:
I can possibly retrieve the entire log of that boot session if you believe that would be helpful.
So far every single boot session gets the endless parity error about an hour in to the boot

dkerr64, 2025-03-15:
I do not need more logs.  The ones you provided are quite clear. I have an idea to try, but I am not at home so will take a few days. 

How is the myQ device connected to the GDO?

dkerr64, 2025-03-15:
Devices connected wirelessly should not interfere with ratgdo.  And the wall panel is expected so again should not be an issue.

EricSihaoLin, 2025-03-15:
> How is the myQ device connected to the GDO?

I have followed the normal setup of this device. bought originally from [Amazon](https://www.amazon.com/Smart-Garage-Opener-Chamberlain-myQ-G0401/dp/B08GD3D9YJ)

The hub communicates to the garage opener most likely via the 310 MHz frequency similar to the keypad as the setup required pressing the purple learn button. The door status is conveyed using the separate smaller device attached to the physical moving door.

The myQ hub then phones home to Chamberlain servers via SSL over Wifi according to packet captures.
This matches the [FCC registration](https://fcc.report/FCC-ID/HBW8522/4251746) for the hub device.

dkerr64, 2025-03-22:
@EricSihaoLin can you try release 3.1.3 please.

- I have made a change to the software serial emulation which might help, so please try that.
- If you are still seeing errors, then I have also added a new option to the settings page.  Where you select the Door Protocol there is a new option "use software serial emulation rather than h/w UART" and it is selected by default.  Try turning this off (so it will use the ESP32's h/w UART) and see if that helps.

Thanks.

EricSihaoLin, 2025-03-27:
I'll report back after flashing this weekend, don't want to leave the software on overnight. Thanks for potentially fixing it.

EricSihaoLin, 2025-03-29:
Unfortunately 3.1.3 release did not help the situation, latest crash logs: https://gist.github.com/EricSihaoLin/d34dfcb5e42be7aa1e34fb25b493804d

What I did notice is that sometimes the ratgdo32 web UI gets unbearably laggy within the first 5 minutes of the device rebooting. No rhyme or reason. The main web ui would have that spinny loading sign up for quite the while.

I know that the ratgdo preferts a 2.4 Ghz wifi, so there's a dedicated SSID only broadcasting that for it.

I'm trying to disable the software serial emulation now.

EricSihaoLin, 2025-03-29:
Yup, even with the software serial emulation turned off, within the first 5 minutes of it powered on after the boot cycle, it crashed and as it crashed it opened the door...

https://gist.github.com/EricSihaoLin/f4cfe3fc62f123c262ab8ca20fabeb53

This log shows the crash, open, and then the subsequent reboot with the open door.
I also closed the door with my physical garage button right after I found it open...

I don't know what to say but to keep this unplugged for now.

dkerr64, 2025-03-29:
> Unfortunately 3.1.3 release did not help the situation, latest crash logs: https://gist.github.com/EricSihaoLin/d34dfcb5e42be7aa1e34fb25b493804d
>
> What I did notice is that sometimes the ratgdo32 web [...]

Notes from looking at log, no conclusions yet.

Things look fine until [line 810](https://gist.github.com/EricSihaoLin/d34dfcb5e42be7aa1e34fb25b493804d#file-gistfile1-txt-L810) where we have a timeout.  From that point on (until reboot) everything is a mess.
At[ line 1728](https://gist.github.com/EricSihaoLin/d34dfcb5e42be7aa1e34fb25b493804d#file-gistfile1-txt-L1738) you request a reboot.
First log message after reboot is at [line 1743](https://gist.github.com/EricSihaoLin/d34dfcb5e42be7aa1e34fb25b493804d#file-gistfile1-txt-L1743). From that point on until end of log file (line 10353) everything looks normal.

dkerr64, 2025-03-29:
> Yup, even with the software serial emulation turned off, within the first 5 minutes of it powered on after the boot cycle, it crashed and as it crashed it opened the door... [...]

Notes from looking at log, no conclusions yet.

- Log starts in error state.  I am assuming that at this point it is running without s/w serial emulation, which indicates that the problem occurs with both h/w UART and s/w serial.
- A reboot occurs at [line 2621](https://gist.github.com/EricSihaoLin/f4cfe3fc62f123c262ab8ca20fabeb53#file-gistfile1-txt-L2621). There are no logs to suggest that this was a user requested reboot, so assume is result of crash.  Is there a traceback in the crashlog web page?
- From that point on until end of log file (line 5956) everything looks normal.

dkerr64, 2025-03-29:
Having looked at both logs,  Things start going bad when a parity error is detected on the data from GDO.  This is occurring with both the h/w UART and s/w serial emulation.   There is no explicit recovery code for parity errors... they are just ignored and we wait for next data.

Based on that...

- Its important to know if this also occurs with the ESP Home firmware.
- It looks like a reboot gets you back working (until parity error occurs again).  Is there something we should do to explicitly try and recover rather than just ignore parity errors (e.g. re-initialize the UART).

@PaulWieland and @descipher FYI and any ideas?

descipher, 2025-03-29:
Based on the very first posted log I believe there is likely a wireline/obstruction contact problem, I would recommend doing a full contact check of all the physical wires. Also I would look for ground faults on wires that are fed through the sharp braces, this is a common issue. Typically an open door command will be completely unaffected but a close will be inconsistent or ignored. An obstruction state (shorted to ground) could trigger an open event if there are enough garbage commands and brown outs on a wall controller.

dkerr64, 2025-03-30:
@EricSihaoLin see comments from @descipher, I think he is probably right that there is a wiring issue.  But I would like to know if the problem exists with ESP Home firmware, that would settle it.

descipher, 2025-03-30:
@dkerr64 Would need a v1 GDO unit to confirm with certainty, I have lots of people on v1's, no reports with v1.1.8 of GDOLIB, based on the behavior we probably need to flush the tx queue on a an upper limit like 15 and 3-5 seconds b4 it goes into the weeds to remain stable under pressure. I will experiment with that. Otherwise it could exception under those conditions.

EricSihaoLin, 2025-04-04:
I’ll also take an entire video of the wiring of the garage door if that can help narrow down the issue as well.

EricSihaoLin, 2025-04-12:
I have flashed the ratgdo32 device to use the ESPHome version of the firmware.
The same oddity has not happened at all with the same exact wiring setup.
The device has generally remained stable and I've been leaving it on for 24 hours for over a week now.

dkerr64, 2025-06-08:
Assuming that problem is solved, will close the issue.

closed 2025-06-08 by dkerr64

dkerr64, 2025-07-13:
@naanlizard If you capture a log that shows the problem occurring then it is probably best to open a new issue so we can track your specific issue.  Thanks.