ratgdo/homekit-ratgdo32 #50: Liftmaster Door TTC set to 5 seconds after update to 3.1.x

ShoGinn, 2025-03-16:
After updating the firmware to 3.1.2, my opener has a hard-set time-to-close (TTC) of 5 seconds.

This means that when I open the door using the wall panel, the remote, or the web interface, the garage door beeps after 5 seconds and starts to close.

I even tried rolling back to the previous version, but the TTC setting seems to have been saved on the opener itself.

Since my wall panel is “dumb,” it doesn’t allow me to modify the TTC time.

I would greatly appreciate any help you can provide.

dkerr64, 2025-03-16:
Can you set the TTC to zero on the settings web page and try again please. 

TTC should only affect control from the webpage and HomeKit. If also affecting wall panel then that is a bug I need to track down and fix.

ShoGinn, 2025-03-16:
I did try setting it to 0. 

No effect. 

I also disconnected the ratgdo and the problem persisted any time the door was opened

ShoGinn, 2025-03-16:
I also so on Reddit this. 

https://www.reddit.com/r/HomeKit/s/rSLU2ja4wO

dkerr64, 2025-03-16:
Ok. Other setting to try is to disable the use built-in TTC. but you may have tried that already.

I am not able to comment in Reddit, if you are able to, then you can comment that author will fix as soon as I get a chance.

Tagging @descipher for his awareness.

dkerr64, 2025-03-16:
BTW… this is nothing to do with wiring or the type of wall panel. I have to fix in my code.

ShoGinn, 2025-03-16:
I figured it was setting some ttc code in the opener itself. Unfortunately a I don't have a wall panel that can configure the auto ttc.

GelidusResearch, 2025-03-17:
@dkerr64 Thanks for the heads up. As a workaround try setting TTC to 300. Depending on whats going on you should get it to stop the 5 second close issue. A TTC of 0 it set right will disable it, just confirmed that here. So it must not be setting it to 0 currently.

jw7519, 2025-03-21:
@dkerr64 Any estimate on when a fix will be out for this?  I'm in the same boat as @ShoGinn no way of configuring the TTC -- that I know of.   Also should you pull 3.1.2 as a release?  I totally installed it *after* this issue was known.

dkerr64, 2025-03-21:
I am back from my travels so should be able to look at this over the weekend.

v3.1.2 was identified as "pre-release" which is the intended method of getting the code out for more widespread testing with out having it automatically appear as an available update (you have to explicitly request pre-releases).  Yes, I know this is kind of a cold response and I do apologize that it is not working properly for you.

jw7519, 2025-03-21:
@dkerr64 I  appreciate you looking into it and reacting quickly.

I don't remember specifically selecting a prerelease, but I'm a newbie on a first install -- I probably didn't read the prompts correctly.

Either way I'm happy to test to make things better.  Just have to deal with a little bit of grief from the Mrs. 🙂

GelidusResearch, 2025-03-21:
@jw7519 try setting TTC to a positive integer of 60 or 300 should work too. I have another person that cleared the closing to quickly with that.

jw7519, 2025-03-21:
Hey @GelidusResearch I'm not exactly sure how to do that.  Can you point me to some documentation?  Changing the Door Close Delay on the Settings webpage does not seem to have an effect.

dkerr64, 2025-03-22:
> Hey [@GelidusResearch](https://github.com/GelidusResearch) I'm not exactly sure how to do that. Can you point me to some documentation? Changing the Door Close Delay on the Settings webpage does not seem to have an effect.

It will have to be done by me in my code.  60 Seconds can be set by sliding the TTC setting all the way to the right.  I don't have a way to set it to 300 right now, but I could add that.

dkerr64, 2025-03-22:
I am looking at the code now.  The intended design is that TTC should only affect webpage / HomeKit requests to close the door.  It should have no effect to how the wall panel button works.  Unfortunately, when you tell the GDO to use it's built-in TTC, that does affect the wall panel button... and one of the intended actions for that is to automatically close the door after it is opened after the TTC delay (why it is designed like that I do not know).

To work around this, when you press the close button on the webpage/homekit it looks to see if TTC is active.  If it is active then it sends a SET_TTC to the door for e.g. 5 seconds, which causes the door to close after the delay.  We then wait for the door to close completely and once it is closed we do another SET_TTC, this time with delay zero.

Delay of zero is supposed to disable TTC on the garage door.  But maybe that is not working on your specific door. @GelidusResearch suggests above that we can disable it by setting to 60 or 300 seconds.  We can try this... 60 is already possible in the code, I will add 300 as an option too, and see if that resets your GDOs.

The lesson from this is that, maybe, relying on the built-in TTC is a bad idea.  I default to that when door protocol is Sec+2.0 but I will revert the default to use our own timer (like we do for Sec+1.0), with warning if a user try to set use of built-in.  That will help for the future, but in the meantime we have to get your doors back to working properly.

More info to come.

PaulWieland, 2025-03-22:
TTC controls shouldn't be incorporated in ratgdo. Aside from issues that can result from users who do not have a TTC capable wall control, you also run a very high risk of a vehicle collision with the door.

https://github.com/ratgdo/esphome-ratgdo/pull/43#issuecomment-2029874121

dkerr64, 2025-03-22:
> TTC controls shouldn't be incorporated in ratgdo. Aside from issues that can result from users who do not have a TTC capable wall control, you also run a very high risk of a vehicle collision with the door.
> 
> [ratgdo/esphome-ratgdo#43 (comment)](https://github.com/ratgdo/esphome-ratgdo/pull/43#issuecomment-2029874121)

I have reached the same conclusion.  In the meantime, I have to get @ShoGinn and @jw7519 doors back to functioning properly.  So have to try setting GDO's TTC to either 0, 60 or 300.

ShoGinn, 2025-03-22:
> TTC controls shouldn't be incorporated in ratgdo. Aside from issues that can result from users who do not have a TTC capable wall control, you also run a very high risk of a vehicle collision with the door.
> 
> https://github.com/ratgdo/esphome-ratgdo/pull/43#issuecomment-2029874121

When mine was broken a breaking of the sensors always stopped the ttc. But I concur the risk is there.

GelidusResearch, 2025-03-22:
There is risk when the wall control does not support TTC since the user would not be aware of it and able to control it from the wall unit. In this case the problem is there are two functions and they current are conflicting. SecplusGDO does not allow anything less than 60s which is the same as the wall control.

dkerr64, 2025-03-22:
I have published release 3.1.3.

- Built-in time-to-close is no longer the default.
- When you change "Use GDO built-in time-to-close" from on to off then I send a SET_TTC to the door.  If the door delay slider is set to less than 60 seconds, then SET_TTC is set to zero... which is supposed to disable TTC in the door.  The slider can also be set to 60 or 300 seconds, if either of those are set then the SET_TTC is set to those values... to see if @GelidusResearch theory is correct.

So @jw7519 please try setting this on, save it.  Then try setting it off again, and save it.  First with a less than 60 seconds set, but if that does not work then try setting to 300 or 60.  Please have "info" level logging on and capture the logs for me.

I consider this a somewhat temporary change in attempt to get @jw7519 and anyone else who gets into this situation to work again (hopefully).  As noted above, it looks like we don't want to ever use the door's built-in TTC.

jw7519, 2025-03-23:
@dkerr64 Works perfectly!

Sorry, I don't think I got the logs working correctly when I first tried it.  But here are the logs I do have.  Let me know if you need me to try anything else:

https://gist.github.com/jw7519/eb65129839aabc3fddd637dc130a7ea3

ss1959, 2025-03-23:
As one of originals to struggle with this issue I thank you for the work on this.
I installed Version 3.1.3 and it works, however Homekit will not actually close the door, opens it fine.
If I go into the Ratgdo settings I can close the door that way. Relays clicks, light flashes once then closes.
Through the Home app I can hear a relay (for lack of a better term) click but doesn't close.

PaulWieland, 2025-03-23:
Are your obstruction sensors connected and working?

ss1959, 2025-03-23:
Yes

dkerr64, 2025-03-23:
Try setting time-to-close to zero and see if that makes a difference.  Also is helpful to capture the logs during the TTC delay.

I have observed HomeKit / Apple Home getting out of sync with door status... but I cannot explain why because when I look at the logs I can see that we are updating HomeKit correctly.

astrocat5, 2025-03-24:
Setting it to zero fixed it for me. I can now close the door from the Home app and Siri. 

I have two garage doors and downgraded one to 3.0.6, it closes fine with 5 seconds delay, flashes a few times then closes.

Just FYI I’m brand new to ratgdo, got them over the weekend, flashed with HomeKit from the website and it installed 3.1.2. Worked briefly then started closing almost immediately as others reported.

dkerr64, 2025-03-30:
Closing this now as v3.1.4 removes option to use GDO's built-in time-to-close.

closed 2025-03-30 by dkerr64