ratgdo/homekit-ratgdo32 #54: v3.1.3 and prior fw, when in Sec+ mode, gdolib TX queue full error

rramondo, 2025-03-29:
New ESP32's arrived. Flashed to 3.1.3 and setup one with a door in sec2.0 mode. Works. Other door is Sec+ mode and the device ends up in some sort of loop.  No ability to add in homekit or do much else.  Downgraded FW to 3.0.6 with same results.  When you switch back to Sec2.0, it works. Switch back to sec+, same result. 

Paste: 
== Setup VL53L4CX time-of-flight sensor ===
I (8.911) gdolib: Using software emulation of serial port instead of h/w UART
I (8.912) ratgdo-comms: generate new id code
I (8.915) ratgdo-comms: id code 15975737 (0xF3C539)
I (8.925) ratgdo-comms: rolling code 0 (0x00)
I (8.927) gdolib: Set serial port protocol to Unknown
I (8.928) gdolib: Set serial port protocol to Security+ 1.0
I (8.928) gdolib: Starting GDO software serial task.
I (8.940) gdolib: GDO Started
I (8.940) gdolib: Set serial port protocol to Security+ 1.0
I (8.940) ratgdo-http: === Starting HTTP web server ===
I (8.950) ratgdo-http: Allocated buffer for JSON, size: 2048
I (8.961) ratgdo-http: Motion triggers, motion : 0, obstruction: 0, light key: 0, door key: 0, lock key: 0, asInt: 0
I (8.971) ratgdo-http: Registering URI handlers
I (8.974) ratgdo-http: Free heap (HTTP server started): 121544
I (8.975) ratgdo-homekit: Status: Need to pair
I (9.175) ratgdo-main: Free heap dropped to 121496
W (9.939) gdolib: Timeout in xTaskNotifyWait
W (10.939) gdolib: Timeout in xTaskNotifyWait
I (11.182) ratgdo-main: Free heap dropped to 121492
W (11.451) gdolib: V1 panel not found, trying emulation
E (11.451) uart: uart_flush_input(1562): uart driver error
W (11.940) gdolib: Timeout in xTaskNotifyWait
I (12.635) ratgdo-http: Client 172.16.33.9 requesting: /index.html (HTTP_GET, type: text/html, length: 6460)
I (12.706) ratgdo-http: Client 172.16.33.9 requesting: /status.json (method: HTTP_GET)
I (12.823) ratgdo-http: JSON length: 1228
I (12.853) ratgdo-http: Client 172.16.33.9 requesting: /rest/events/subscribe (method: HTTP_GET)
I (12.856) ratgdo-http: SSE Subscription for client 4d06d41c-f276-43ad-8708-eb6de342abe9 with IP 172.16.33.9: event bus location: /rest/events/0, Total subscribed: 1
I (12.924) ratgdo-http: Client 172.16.33.9 listening for SSE events on channel 0
W (12.941) gdolib: Timeout in xTaskNotifyWait
I (13.184) ratgdo-main: Free heap dropped to 118488
W (13.942) gdolib: Timeout in xTaskNotifyWait
I (14.193) ratgdo-main: Free heap dropped to 118176
W (14.943) gdolib: Timeout in xTaskNotifyWait
W (15.944) gdolib: Timeout in xTaskNotifyWait
I (16.402) ratgdo-http: Client 172.16.33.9 requesting: /logs.html (HTTP_GET, type: text/html, length: 1498)
I (16.448) ratgdo-http: Client 172.16.33.9 requesting: /status.json (method: HTTP_GET)
I (16.452) ratgdo-comms: GDO event: synced: true, protocol: Security+ 1.0
I (16.568) ratgdo-http: JSON length: 1230
I (16.598) ratgdo-http: Client 172.16.33.9 requesting: /rest/events/subscribe (method: HTTP_GET)
I (16.600) ratgdo-http: SSE Subscription for client d9a09002-9d88-44e8-8f65-d7b72dd208db with IP 172.16.33.9: event bus location: /rest/events/1, Total subscribed: 2
I (16.643) ratgdo-http: Client 172.16.33.9 requesting: /showrebootlog (method: HTTP_GET)
I (16.647) ratgdo-logger: Print saved log
I (16.677) ratgdo-http: Client 172.16.33.9 requesting: /crashlog (method: HTTP_GET)
I (16.678) ratgdo-http: Request to display crash log...
I (16.722) ratgdo-http: Client 172.16.33.9 requesting: /showlog (method: HTTP_GET)
I (17.445) ratgdo-http: Client 172.16.33.9 listening for SSE events on channel 1
W (17.946) gdolib: Timeout in xTaskNotifyWait
W (18.950) gdolib: Timeout in xTaskNotifyWait
I (19.241) ratgdo-main: Free heap dropped to 117868
W (19.954) gdolib: Timeout in xTaskNotifyWait
I (20.242) ratgdo-main: Free heap dropped to 117420
E (20.452) gdolib: gdo TX queue full!
E (20.702) gdolib: gdo TX queue full!
E (20.952) gdolib: gdo TX queue full!
W (20.958) gdolib: Timeout in xTaskNotifyWait
E (21.202) gdolib: gdo TX queue full!
I (21.247) ratgdo-main: Free heap dropped to 115708
E (21.452) gdolib: gdo TX queue full!
E (21.702) gdolib: gdo TX queue full!
E (21.952) gdolib: gdo TX queue full!
W (21.962) gdolib: Timeout in xTaskNotifyWait
E (22.202) gdolib: gdo TX queue full!
I (22.249) ratgdo-main: Free heap dropped to 115272
E (22.452) gdolib: gdo TX queue full!
E (22.702) gdolib: gdo TX queue full!
E (22.952) gdolib: gdo TX queue full!
[... 108 lines cut ...]
E (42.452) gdolib: gdo TX queue full!
E (42.702) gdolib: gdo TX queue full!
E (42.952) gdolib: gdo TX queue full!
W (43.047) gdolib: Timeout in xTaskNotifyWait
E (43.202) gdolib: gdo TX queue full!
E (43.452) gdolib: gdo TX queue full!
E (43.702) gdolib: gdo TX queue full!
E (43.952) gdolib: gdo TX queue full!
W (44.051) gdolib: Timeout in xTaskNotifyWait
E (44.202) gdolib: gdo TX queue full!
E (44.453) gdolib: gdo TX queue full!
E (44.702) gdolib: gdo TX queue full!
E (44.952) gdolib: gdo TX queue full!
W (45.055) gdolib: Timeout in xTaskNotifyWait
E (45.202) gdolib: gdo TX queue full!
E (45.453) gdolib: gdo TX queue full!
E (45.702) gdolib: gdo TX queue full!
E (45.953) gdolib: gdo TX queue full!
W (46.059) gdolib: Timeout in xTaskNotifyWait
E (46.202) gdolib: gdo TX queue full!
E (46.452) gdolib: gdo TX queue full!
E (46.702) gdolib: gdo TX queue full!
E (46.952) gdolib: gdo TX queue full!
For documentation and support see the [GitHub](https://github.com/ratgdo/homekit-ratgdo32) page.
Copyright (c) 2024-25 [homekit-ratgdo contributors.](https://github.com/ratgdo/homekit-ratgdo32/graphs/contributors)

dkerr64, 2025-03-30:
Can you set debug for the log-level in settings please, and run again to capture more detail please.  I am about to publish a 3.1.4 release so use that please (though there is nothing changed in that likely to affect whatever you are seeing).

rramondo, 2025-03-30:
I had to take the board down from the door and plug it into my PC, so there are no external attached in's/out's.
Debug: 
Current server message log...

Server upTime:       0 days 00 hrs 00 mins 07 secs
Server uptime (secs): 7.693
Firmware version: 3.1.3
Free heap: 115252
Minimum heap: 115252

I (12) ratgdo-homekit: === Setup HomeKit accessories and services ===
I (13) ratgdo-homekit: HomeKit pairing QR Code ID: 4xui
I (13) ratgdo-homekit: HomeKit QR setup payload: X-HM://0042WZMX34xui
I (2.218) ratgdo-homekit: Configuring HomeKit Garage Door Service
I (2.219) ratgdo-homekit: Configuring HomeKit Light Service for GDO Light
I (2.230) ratgdo-homekit: No motion sensor. Skipping motion service
I (2.231) ratgdo-main: Free heap dropped to 199400
I (5.381) ratgdo-homekit: Status: WiFi connecting
I (5.392) ratgdo-homekit: Wifi begin for SSID: GBC-Iot
I (6.155) ratgdo-main: Free heap dropped to 151356
I (7.156) ratgdo-main: Free heap dropped to 151340
I (8.157) ratgdo-main: Free heap dropped to 151028
I (8.301) ratgdo-homekit: WiFi established, count: 1, IP: 172.16.36.136, Mask: 255.255.255.0, Gateway: 172.16.36.1, DNS: 172.16.32.2
I (8.316) ratgdo-vehicle: === Setup VL53L4CX time-of-flight sensor ===
I (8.444) gdolib: Using software emulation of serial port instead of h/w UART
I (8.446) ratgdo-comms: generate new id code
I (8.449) ratgdo-comms: id code 16237881 (0xF7C539)
I (8.450) ratgdo-comms: rolling code 0 (0x00)
I (8.462) gdolib: Set serial port protocol to Unknown
I (8.463) gdolib: Set serial port protocol to Security+ 1.0
I (8.463) gdolib: Starting GDO software serial task.
I (8.475) gdolib: GDO Started
I (8.475) gdolib: Set serial port protocol to Security+ 1.0
I (8.476) ratgdo-http: === Starting HTTP web server ===
I (8.487) ratgdo-http: Allocated buffer for JSON, size: 2048
I (8.488) ratgdo-http: Motion triggers, motion : 0, obstruction: 0, light key: 0, door key: 0, lock key: 0, asInt: 0
I (8.499) ratgdo-http: Registering URI handlers
I (8.514) ratgdo-http: Free heap (HTTP server started): 121272
I (8.516) ratgdo-homekit: Status: Need to pair
I (9.170) ratgdo-main: Free heap dropped to 121236
W (9.474) gdolib: Timeout in xTaskNotifyWait
I (10.168) ratgdo-http: Client 172.16.33.9 requesting: /logs.html (HTTP_GET, type: text/html, length: 1498)
I (10.196) ratgdo-main: Free heap dropped to 115252
I (10.311) ratgdo-http: Client 172.16.33.9 requesting: /status.json (method: HTTP_GET)
I (10.429) ratgdo-http: JSON length: 1226
I (10.457) ratgdo-http: Client 172.16.33.9 requesting: /rest/events/subscribe (method: HTTP_GET)
I (10.460) ratgdo-http: SSE Subscription for client f2b5edeb-7aba-44dc-aa4a-0827b4fd2c44 with IP 172.16.33.9: event bus location: /rest/events/0, Total subscribed: 1
W (10.474) gdolib: Timeout in xTaskNotifyWait
I (10.506) ratgdo-http: Client 172.16.33.9 requesting: /showrebootlog (method: HTTP_GET)
I (10.510) ratgdo-logger: Print saved log
I (10.539) ratgdo-http: Client 172.16.33.9 requesting: /crashlog (method: HTTP_GET)
I (10.540) ratgdo-http: Request to display crash log...
I (10.582) ratgdo-http: Client 172.16.33.9 requesting: /showlog (method: HTTP_GET)
I (11.306) ratgdo-http: Client 172.16.33.9 listening for SSE events on channel 0
W (11.475) gdolib: Timeout in xTaskNotifyWait
D (11.483) gdolib: TX [35]
D (11.489) gdolib: Collision detected, ignoring as in middle of sync
D (11.730) gdolib: TX [35]
D (11.737) gdolib: Collision detected, ignoring as in middle of sync
D (11.979) gdolib: TX [35]
D (11.984) gdolib: Collision detected, ignoring as in middle of sync
D (12.229) gdolib: TX [33]
D (12.235) gdolib: Collision detected, ignoring as in middle of sync
W (12.479) gdolib: Timeout in xTaskNotifyWait
D (12.479) gdolib: TX [33]
D (12.490) gdolib: Collision detected, ignoring as in middle of sync
D (12.729) gdolib: TX [53]
D (12.734) gdolib: Collision detected, ignoring as in middle of sync
D (12.979) gdolib: TX [53]
D (12.985) gdolib: Collision detected, ignoring as in middle of sync
[... 692 lines cut ...]
D (23.752) gdolib: Collision detected, requeuing command
D (23.757) gdolib: Collision detected, requeuing command
D (23.769) gdolib: Collision detected, requeuing command
D (23.803) gdolib: Collision detected, requeuing command
D (23.808) gdolib: Collision detected, requeuing command
D (23.814) gdolib: Collision detected, requeuing command
D (23.821) gdolib: Collision detected, requeuing command
D (23.833) gdolib: Collision detected, requeuing command
D (23.838) gdolib: Collision detected, requeuing command
D (23.842) gdolib: Collision detected, requeuing command
D (23.862) gdolib: Collision detected, requeuing command
D (23.867) gdolib: Collision detected, requeuing command
D (23.870) gdolib: Collision detected, requeuing command
D (23.896) gdolib: Collision detected, requeuing command
D (23.900) gdolib: Collision detected, requeuing command
D (23.903) gdolib: Collision detected, requeuing command
D (23.907) gdolib: Collision detected, requeuing command
D (23.910) gdolib: Collision detected, requeuing command
D (23.923) gdolib: Collision detected, requeuing command
D (23.957) gdolib: Collision detected, requeuing command
D (23.963) gdolib: Collision detected, requeuing command
D (23.967) gdolib: Collision detected, requeuing command
D (23.974) gdolib: Collision detected, requeuing command
D (23.979) gdolib: TX [38]
E (23.984) gdolib: gdo TX queue full!

rramondo, 2025-03-30:
Flashed to 3.1.4 and reinstalled to the opener.  Noticed the Red wire to the opener from the 32disco was not fully seated. Now it is. The device is not able to read the door status accurately? And I can't add it to home kit now? Times out from my apple iphone when trying to add to home?  Debug Below:
Current server message log...

Server upTime:       0 days 00 hrs 00 mins 09 secs
Server uptime (secs): 9.822
Firmware version: 3.1.4
Free heap: 86092
Minimum heap: 86092

3A 58]
D (10.993) gdolib: Light state: Off
D (10.993) gdolib: Lock state: Unlocked
D (11.004) gdo_utils: RX [39 00]
D (11.004) gdolib: Obstruction state: Clear
D (11.004) gdo_utils: RX [37 00]
D (11.004) gdo_utils: TX [3A]
I (11.015) ratgdo-comms: GDO event: light: Off
I (11.015) ratgdo-comms: GDO event: lock: Unlocked
I (11.015) ratgdo-comms: GDO event: obstruction: Clear
D (11.035) gdolib: Sent command: QUERY_OTHER_STATUS
D (11.043) ratgdo-vehicle: Unusual VL53L4CX Range Status: 12, Range: 1631
D (11.430) ratgdo-vehicle: Unusual VL53L4CX Range Status: 12, Range: 1616
D (11.495) ratgdo-vehicle: Unusual VL53L4CX Range Status: 12, Range: 1630
D (11.562) ratgdo-vehicle: Unusual VL53L4CX Range Status: 12, Range: 1628
D (11.627) ratgdo-vehicle: Unusual VL53L4CX Range Status: 12, Range: 1628
D (11.692) ratgdo-vehicle: Unusual VL53L4CX Range Status: 12, Range: 1619
D (11.759) ratgdo-vehicle: Unusual VL53L4CX Range Status: 12, Range: 1597
D (11.890) ratgdo-vehicle: Unusual VL53L4CX Range Status: 12, Range: 1609
W (12.036) sw_serial: Timeout in xTaskNotifyWait
W (12.037) sw_serial: Parity error 0xFC
D (12.037) gdo_utils: RX [3A 58]
D (12.037) gdolib: Light state: Off
D (12.048) gdolib: Lock state: Unlocked
D (12.048) gdo_utils: RX [3A 58]
D (12.048) gdolib: Light state: Off
D (12.048) gdolib: Lock state: Unlocked
D (12.219) ratgdo-vehicle: Unusual VL53L4CX Range Status: 12, Range: 1621
D (12.286) ratgdo-vehicle: Unusual VL53L4CX Range Status: 12, Range: 1609
I (12.334) ratgdo-http: Client 172.16.33.9 requesting: /logs.html (HTTP_GET, type: text/html, length: 1498)
I (12.397) ratgdo-http: Client 172.16.33.9 requesting: /status.json (method: HTTP_GET)
I (12.516) ratgdo-http: JSON length: 1226
W (12.525) gdolib: V1 panel not found, trying emulation
D (12.532) ratgdo-vehicle: Unusual VL53L4CX Range Status: 12, Range: 1606
I (12.541) ratgdo-main: Free heap dropped to 86092
I (12.549) ratgdo-http: Client 172.16.33.9 requesting: /rest/events/subscribe (method: HTTP_GET)
I (12.552) ratgdo-http: SSE Subscription for client eff62f78-4224-4c55-abf8-d2b6c5a93688 with IP 172.16.33.9: event bus location: /rest/events/0, Total subscribed: 1
I (12.598) ratgdo-http: Client 172.16.33.9 requesting: /showrebootlog (method: HTTP_GET)
I (12.602) ratgdo-logger: Print saved log
D (12.620) ratgdo-vehicle: Unusual VL53L4CX Range Status: 12, Range: 1606
I (12.637) ratgdo-http: Client 172.16.33.9 requesting: /crashlog (method: HTTP_GET)
I (12.638) ratgdo-http: Request to display crash log...
D (12.778) gdo_utils: TX [35]
D (12.790) gdolib: Sent command: TOGGLE_LOCK_RELEASE
D (12.791) gdo_utils: RX [39 00]
D (12.792) gdolib: Obstruction state: Clear
D (12.792) gdo_utils: RX [37 00]
D (12.802) gdo_utils: TX [3A]
D (12.803) gdolib: TX pending, waiting, 12ms since last TX
I (12.815) ratgdo-http: Client 172.16.33.9 requesting: /index.html (HTTP_GET, type: text/html, length: 6461)
D (12.851) gdolib: Sent command: QUERY_OTHER_STATUS
W (12.851) sw_serial: Parity error 0x32
I (12.923) ratgdo-http: Client 172.16.33.9 requesting: /showlog (method: HTTP_GET)
I (13.120) ratgdo-http: Client 172.16.33.9 listening for SSE events on channel 0
D (13.276) gdo_utils: TX [35]
D (13.292) gdolib: Sent command: TOGGLE_LOCK_RELEASE
D (13.296) gdo_utils: RX [3A 58]
D (13.300) gdolib: Light state: Off
D (13.303) gdolib: Lock state: Unlocked
I (13.383) ratgdo-http: Client 172.16.33.9 requesting: /status.json (method: HTTP_GET)
I (13.504) ratgdo-http: JSON length: 1228
[... 476 lines cut ...]
D (32.799) gdolib: Sent command: OBSTRUCTION
D (32.802) gdo_utils: RX [37 00]
D (32.805) gdo_utils: TX [3A]
D (32.808) gdolib: TX pending, waiting, 6ms since last TX
W (32.864) sw_serial: Parity error 0x32
D (32.865) gdolib: Sent command: QUERY_OTHER_STATUS
D (32.872) gdo_utils: RX [39 00]
D (32.874) gdolib: Obstruction state: Clear
D (32.999) ratgdo-vehicle: Unusual VL53L4CX Range Status: 12, Range: 1653
D (33.026) gdo_utils: TX [38]
W (33.036) sw_serial: Parity error 0xFC
D (33.050) gdolib: Sent command: QUERY_DOOR_STATUS
D (33.069) ratgdo-vehicle: Unusual VL53L4CX Range Status: 12, Range: 1650
D (33.264) ratgdo-vehicle: Unusual VL53L4CX Range Status: 12, Range: 1633
D (33.276) gdo_utils: TX [3A]
W (33.292) sw_serial: Parity error 0x32
D (33.293) gdolib: Sent command: QUERY_OTHER_STATUS
D (33.299) gdo_utils: RX [38 53]
D (33.302) gdolib: Door state: Unknown
I (33.304) ratgdo-comms: GDO event: door: Unknown,   0%, target:  -1%
I (33.326) ratgdo-http: Current Door State changing from Opening to Unknown
I (33.331) ratgdo-http: SSE send to client 172.16.33.9 on channel 0, data: { "lastDoorUpdateAt": 0, "garageDoorState": "Unknown", "upTime": 30221 }
I (33.355) ratgdo-http: SSE send to client 172.16.33.9 on channel 1, data: { "lastDoorUpdateAt": 0, "garageDoorState": "Unknown", "upTime": 30221 }
D (33.384) ratgdo-vehicle: Unusual VL53L4CX Range Status: 12, Range: 1647
D (33.526) gdo_utils: TX [39]

rramondo, 2025-03-30:
The Opener is a liftmaster 3800 series w/ one remote opener wall switch.  The Ratgdo32-Disco is connected like the older ESP16 based ratgdo was.   The status page of the Ratdgo32Disco shows the "door state" flopping from Opening to Unknown to Open to Stopped and back around again.  I can't get the 32 to add to homekit. Device Times out is all I get. From the web interface I can turn the light on/off properly.  The open/close button will open the door from closed state, but takes a few requests (4) to actually close the door from an open state.  When the door is open, it reports properly Door State: Open.  Obstruction reports properly false/true if the beam if unblocked or blocked. Any Clue as to why it will not respond to home kit?  Home WiFi is a Small Office / Enterprise oriented Ubiquity network w/ a mix of U7/U6 AP's.  SSID the device is connected to has about 40 other homekit nodes on that specific ssid setup for IoT things. Thanks!

rramondo, 2025-03-30:
Last update for today.  Noticed that the user is required to be on the same network as the Bridge Device as it is implemented today.  Other accessories can deal with the mDNS / broadcast discovery methods.  I have not had a chance to look thru the code to see if you were able to implement this as part of the bridge code.   If I moved the iPhone over to the IoT SSID/Network, I was able to add the bridge and all the subsequent accessories!  yay.  Still have the door status issues and as such it is throwing a ton of Door Open alerts.  Let me know if I can do any further logging/explaining. Thank you!

rramondo, 2025-03-30:
The second Ratgdo32Disco installed on the other garage door (Sec 2.0 compliant) is working without a hitch. (minus the home kid adding accessory thing).

dkerr64, 2025-03-31:
Your second log shows that things are sort-of working.  But there are an awful lot of parity errors and this would explain why some of the requests to open or close the door are not working.  check your wiring and make sure that you have solid connections.

Also, to reduce the amount of traffic to the door on a close request, set the time-to-close delay to zero.

And you could consider disabling the s/w serial emulation (so it uses the h/w UART) but I still expect you will see a lot of parity errors until you sort out the wiring connections.

dkerr64, 2025-03-31:
And just to comment on your log... the unusual VL53L4CX range status of 12... could in fact be fairly normal.  I see it a lot when my car is not parked in the garage, but never when the car is there.  The distance reading is used.  Maybe I shouldn't log these (as debug) if they are going to be so common.  Maybe I move to verbose.

dkerr64, 2025-03-31:
As for mDNS discovery.  That is outside the control of the firmware.  It responds to mDNS requests that it receives.  If you are connecting the ratgdo to a VLAN then you need to make sure that you are "reflecting" the mDNS requests across your different subnets.  I do this myself, my ratgdos are on a VLAN (with other smart devices) isolated from my main network.  Apple Home hubs and my phones/computers are on a separate VLAN.  I have avahi installed on a bridge that reflects the mDNS traffic across each subnet.  I have no problems pairing or communicating with the ratgdo's.

dkerr64, 2025-06-08:
@rramondo Are you still having problems?

rramondo, 2025-06-12:
Ah, yea. Thx for the followup.  This garage door interface is not working at all and I've stopped trying.  The new-er garage door opener on the other garage works as expected with the Ratgdo device.  At this point, it would probably take replacing / upgrading the older unit.  I'll suffer until then.

dkerr64, 2025-06-15:
Thanks for the update.  I will close this issue then.

closed 2025-06-15 by dkerr64