ratgdo/homekit-ratgdo32 #57: Wall switch doesn't work after updating dry contact to v3.1.4

aberkvam, 2025-03-31:
Updated to 3.1.4.  The wall switch stopped working.  Pressing it doesn't do anything.  Nothing shows up in the debug logs when I press the wall switch.  HomeKit and radio controls still work fine.  Rebooting doesn't get the wall switch to work.

```
{
"upTime": 1986250,
"deviceName": "Garage Door",
"userName": "admin",
"paired": true,
"firmwareVersion": "3.1.4",
"localIP": "10.0.1.112",
"subnetMask": "255.255.255.0",
"gatewayIP": "10.0.1.1",
"nameserverIP": "8.8.8.8",
"macAddress": "1C:69:20:C2:C0:1C",
"wifiSSID": "XXXXX",
"wifiRSSI": "-50 dBm, Channel 5",
"wifiBSSID": "XXXXX",
"lockedAP": false,
"GDOSecurityType": 3,
"garageDoorState": "Closed",
"garageLockState": "Unknown",
"garageLightOn": false,
"garageMotion": false,
"garageObstructed": false,
"passwordRequired": false,
"rebootSeconds": 0,
"freeHeap": 106444,
"minHeap": 93492,
"crashCount": 9738,
"staticIP": false,
"syslogEn": false,
"syslogIP": "0.0.0.0",
"syslogPort": 514,
"logLevel": 4,
"TTCseconds": 2,
"builtInTTC": false,
"vehicleThreshold": 100,
"motionTriggers": 0,
"LEDidle": 0,
"lastDoorUpdateAt": 1995990,
"enableNTP": true,
"serverTime": 1743400377,
"timeZone": "America/Chicago;CST6CDT,M3.2.0,M11.1.0",
"distanceSensor": false,
"laserEnabled": false,
"laserHomeKit": true,
"vehicleHomeKit": false,
"dcOpenClose": false,
"useSWserial": true,
"dcDebounceDuration": 50,
"assistDuration": 60,
"qrPayload": "XXXXX"
}
```

dkerr64, 2025-03-31:
What version were you on previously?  Last time dry contact code was touched was in 3.1.1, no changes since.

aberkvam, 2025-03-31:
I was on 3.1.2. The wall button worked fine with that version.  

I’ll do some more detailed testing later today (roll back to previous versions, look at start up debug logs carefully, etc.).

aberkvam, 2025-03-31:
Not sure if this is related.  I tried to open the door for the first time today.  Nothing worked.  HomeKit just said the door was “Opening” but nothing was happening.  The RATGDO web GUI let me click on the “Open Door” button but nothing happened.  The wall switch doesn’t work either.  The logs are filled with this:

```
D (46792.581) gdolib: Collision detected, requeuing command
D (46792.587) gdolib: Collision detected, requeuing command
D (46792.603) gdolib: Collision detected, requeuing command
D (46792.614) gdolib: Collision detected, requeuing command
D (46792.631) gdolib: Collision detected, requeuing command
D (46792.635) gdolib: Collision detected, requeuing command
D (46792.645) gdolib: Collision detected, requeuing command
D (46792.729) gdolib: Collision detected, requeuing command
D (46792.734) gdolib: Collision detected, requeuing command
D (46792.740) gdolib: Collision detected, requeuing command
D (46792.757) gdolib: Collision detected, requeuing command
D (46792.767) gdolib: Collision detected, requeuing command
D (46792.785) gdolib: Collision detected, requeuing command
D (46792.789) gdolib: Collision detected, requeuing command
D (46792.798) gdolib: Collision detected, requeuing command
D (46792.883) gdolib: Collision detected, requeuing command
D (46792.888) gdolib: Collision detected, requeuing command
D (46792.894) gdolib: Collision detected, requeuing command
D (46792.910) gdolib: Collision detected, requeuing command
D (46792.921) gdolib: Collision detected, requeuing command
D (46792.938) gdolib: Collision detected, requeuing command
D (46792.942) gdolib: Collision detected, requeuing command
D (46792.951) gdolib: Collision detected, requeuing command
D (46793.036) gdolib: Collision detected, requeuing command
D (46793.041) gdolib: Collision detected, requeuing command
D (46793.047) gdolib: Collision detected, requeuing command
D (46793.064) gdolib: Collision detected, requeuing command
D (46793.074) gdolib: Collision detected, requeuing command
```

I rebooted the RATGDO and tried opening it via HomeKit.  Still didn’t work.  Here are those logs:

```
Server uptime (secs): 11.471
Firmware version: 3.1.4
Free heap: 104496
Minimum heap: 104496

(9.810) ratgdo-vehicle: VL53L4CX failed to initialize error: 4
I (9.822) gdolib: Initialize dry contact ISR and Task for pin: 13 (open sensor)
I (9.822) gdolib: GDO Started
I (9.823) ratgdo-http: === Starting HTTP web server ===
I (9.823) gdolib: Initialize dry contact ISR and Task for pin: 14 (closed sensor)
D (9.834) gdolib: Wake up for GPIO pin 14 (closed sensor), level is 0
D (9.823) gdolib: Wake up for GPIO pin 13 (open sensor), level is 1
I (9.834) ratgdo-http: Allocated buffer for JSON, size: 2048
I (9.855) ratgdo-http: Motion triggers, motion : 0, obstruction: 0, light key: 0, door key: 0, lock key: 0, asInt: 0
I (9.865) ratgdo-http: Registering URI handlers
I (9.867) ratgdo-http: Free heap (HTTP server started): 117004
I (9.879) ratgdo-homekit: Status: Paired
D (9.894) gdolib: Level of pin 14 (closed sensor) is 0. Interrupts during debounce: 0 (0.000ms)
D (9.895) gdolib: Close sensor pressed
D (9.905) gdolib: Door state: Closed
D (9.906) gdolib: Level of pin 13 (open sensor) is 1. Interrupts during debounce: 0 (0.000ms)
I (9.906) ratgdo-comms: GDO event: door: Closed, 100%, target:  -1%
D (9.916) gdolib: Open sensor released
I (9.917) ratgdo-http: Current Door State changing from Unknown to Closed
D (9.917) gdolib: Door state: Closed
D (9.929) gdolib: Obstruction state: Clear
I (9.939) ratgdo-comms: GDO event: obstruction: Clear
I (9.940) ratgdo-homekit: Garage door set CurrentDoorState: 1
I (9.941) ratgdo-homekit: Garage door set ObstructionDetected: 0
I (10.201) ratgdo-main: Free heap dropped to 116940
I (12.694) ratgdo-http: Client 10.0.1.103 requesting: /index.html (HTTP_GET, type: text/html, length: 6461)
I (13.033) ratgdo-http: Client 10.0.1.103 requesting: /status.json (method: HTTP_GET)
I (13.139) ratgdo-http: JSON length: 1142
I (13.204) ratgdo-main: Free heap dropped to 114792
[... 29 lines cut ...]
D (43.895) gdolib: Collision detected, requeuing command
D (43.908) gdolib: Collision detected, requeuing command
D (44.048) gdolib: Collision detected, requeuing command
D (44.061) gdolib: Collision detected, requeuing command
D (44.201) gdolib: Collision detected, requeuing command
D (44.216) gdolib: Collision detected, requeuing command
D (44.242) gdo_utils: Dry Contact set pin 26 to 0
D (44.250) gdolib: Collision detected, requeuing command
D (44.257) gdo_utils: Dry Contact set pin 17 to 0
D (44.263) gdolib: Collision detected, requeuing command
D (44.355) gdolib: Collision detected, requeuing command
D (44.369) gdolib: Collision detected, requeuing command
D (44.404) gdolib: Collision detected, requeuing command
D (44.418) gdolib: Collision detected, requeuing command
D (44.508) gdolib: Collision detected, requeuing command
D (44.521) gdolib: Collision detected, requeuing command
D (44.557) gdolib: Collision detected, requeuing command
D (44.571) gdolib: Collision detected, requeuing command
D (44.661) gdolib: Collision detected, requeuing command
D (44.675) gdolib: Collision detected, requeuing command
D (44.710) gdolib: Collision detected, requeuing command
D (44.725) gdolib: Collision detected, requeuing command
D (44.814) gdolib: Collision detected, requeuing command
D (44.828) gdolib: Collision detected, requeuing command
```

aberkvam, 2025-03-31:
I downgraded to v3.1.3.  The behavior was the same.  I downgraded to 3.1.2 and things worked.

aberkvam, 2025-03-31:
It's an improvement (wall switch seems to work), but there are some issues/regressions.

First, the simple one.  The version reports as "Firmware:	v3.1.4 Update available (v3.1.5)" even if I download https://github.com/ratgdo/homekit-ratgdo32/releases/download/v3.1.5/homekit-ratgdo32-v3.1.5.firmware.bin directly and upload it to the RATGDO.

Second, the garage door state isn't correctly being relayed to HomeKit.  Right now the door is closed but HomeKit says it's "Opening...".  Tapping the button once in HomeKit changes the status to "Closed" without the door doing anything. The related logs are:
```
I (392.464) ratgdo-homekit: Garage Door Characteristics Update
I (392.470) ratgdo-comms: Door already closed; ignored request
```
At that point, opening the door seems to work correctly.  I hit the button in HomeKit and get:
```
I (466.908) ratgdo-comms: Opening door
D (466.916) gdo_utils: Dry Contact set pin 26 to 1
D (466.925) gdo_utils: Dry Contact set pin 17 to 1
D (467.011) gdolib: Wake up for GPIO pin 13 (open sensor), level is 1
D (467.011) gdolib: Wake up for GPIO pin 14 (closed sensor), level is 0
D (467.064) gdolib: Level of pin 13 (open sensor) is 1. Interrupts during debounce: 0 (0.000ms)
D (467.070) gdolib: Open sensor released
D (467.075) gdolib: Door state: Closed
I (467.082) ratgdo-comms: GDO event: door: Closed, 100%, target:   0%
D (467.082) gdolib: Level of pin 14 (closed sensor) is 0. Interrupts during debounce: 0 (0.000ms)
D (467.095) gdolib: Close sensor pressed
D (467.099) gdolib: Door state: Closed
D (467.425) gdo_utils: Dry Contact set pin 26 to 0
D (467.432) gdo_utils: Dry Contact set pin 17 to 0
D (467.581) gdolib: Wake up for GPIO pin 14 (closed sensor), level is 1
D (467.634) gdolib: Level of pin 14 (closed sensor) is 1. Interrupts during debounce: 0 (0.000ms)
D (467.640) gdolib: Close sensor released
D (467.645) gdolib: Door state: Opening
I (467.652) ratgdo-comms: GDO event: door: Opening, 100%, target:   0%
I (467.657) ratgdo-http: Current Door State changing from Closed to Opening
I (467.660) ratgdo-homekit: Garage door set TargetDoorState: 0
I (467.666) ratgdo-http: SSE send to client 10.0.1.108 on channel 0, data: { "lastDoorUpdateAt": 0, "garageDoorState": "Opening", "upTime": 464659 }
I (467.681) ratgdo-homekit: Garage door set CurrentDoorState: 2
I (467.687) ratgdo-http: SSE send to client 10.0.1.108 on channel 1, data: { "lastDoorUpdateAt": 0, "garageDoorState": "Opening", "upTime": 464659 }
I (467.711) ratgdo-http: SSE send to client 10.0.1.108 on channel 2, data: { "lastDoorUpdateAt": 0, "garageDoorState": "Opening", "upTime": 464659 }
D (467.717) gdolib: Wake up for GPIO pin 14 (closed sensor), level is 0
D (467.780) gdolib: Level of pin 14 (closed sensor) is 0. Interrupts during debounce: 2 (0.029ms)
D (467.786) gdolib: Close sensor pressed
D (467.792) gdolib: Door state: Closed
I (467.799) ratgdo-comms: GDO event: door: Closed, 100%, target:   0%
I (467.803) ratgdo-http: Current Door State changing from Opening to Closed
I (467.808) ratgdo-http: SSE send to client 10.0.1.108 on channel 0, data: { "lastDoorUpdateAt": 0, "garageDoorState": "Closed", "upTime": 464805 }
I (467.820) ratgdo-homekit: Garage door set CurrentDoorState: 1
I (467.824) ratgdo-http: SSE send to client 10.0.1.108 on channel 1, data: { "lastDoorUpdateAt": 0, "garageDoorState": "Closed", "upTime": 464805 }
I (467.844) ratgdo-http: SSE send to client 10.0.1.108 on channel 2, data: { "lastDoorUpdateAt": 0, "garageDoorState": "Closed", "upTime": 464805 }
I (467.861) ratgdo-main: Free heap dropped to 83072
D (467.881) gdolib: Wake up for GPIO pin 14 (closed sensor), level is 1
D (467.934) gdolib: Level of pin 14 (closed sensor) is 1. Interrupts during debounce: 0 (0.000ms)
D (467.940) gdolib: Close sensor released
D (467.945) gdolib: Door state: Opening
I (467.952) ratgdo-comms: GDO event: door: Opening, 100%, target:   0%
I (467.956) ratgdo-http: Current Door State changing from Closed to Opening
I (467.962) ratgdo-http: SSE send to client 10.0.1.108 on channel 0, data: { "lastDoorUpdateAt": 0, "garageDoorState": "Opening", "upTime": 464958 }
I (467.974) ratgdo-homekit: Garage door set TargetDoorState: 0
I (467.978) ratgdo-http: SSE send to client 10.0.1.108 on channel 1, data: { "lastDoorUpdateAt": 0, "garageDoorState": "Opening", "upTime": 464958 }
I (467.997) ratgdo-http: SSE send to client 10.0.1.108 on channel 2, data: { "lastDoorUpdateAt": 0, "garageDoorState": "Opening", "upTime": 464958 }
I (468.009) ratgdo-homekit: Garage door set CurrentDoorState: 2
I (468.452) ratgdo-comms: GDO event: door: Opening,  87%, target:   0%
I (468.952) ratgdo-comms: GDO event: door: Opening,  83%, target:   0%
I (469.452) ratgdo-comms: GDO event: door: Opening,  78%, target:   0%
I (469.952) ratgdo-comms: GDO event: door: Opening,  74%, target:   0%
I (470.452) ratgdo-comms: GDO event: door: Opening,  70%, target:   0%
I (470.952) ratgdo-comms: GDO event: door: Opening,  66%, target:   0%
I (471.452) ratgdo-comms: GDO event: door: Opening,  61%, target:   0%
I (471.952) ratgdo-comms: GDO event: door: Opening,  57%, target:   0%
I (472.452) ratgdo-comms: GDO event: door: Opening,  53%, target:   0%
I (472.952) ratgdo-comms: GDO event: door: Opening,  49%, target:   0%
I (473.452) ratgdo-comms: GDO event: door: Opening,  45%, target:   0%
I (473.952) ratgdo-comms: GDO event: door: Opening,  40%, target:   0%
[... 90 lines cut ...]
I (551.333) ratgdo-http: Current Door State changing from Closing to Closed
[... cut ...]
I (551.543) ratgdo-homekit: Garage door set CurrentDoorState: 1
I (551.545) ratgdo-http: SSE send to client 10.0.1.108 on channel 0, data: { "lastDoorUpdateAt": 0, "garageDoorState": "Closed", "upTime": 548542 }
I (551.568) ratgdo-http: SSE send to client 10.0.1.108 on channel 1, data: { "lastDoorUpdateAt": 0, "garageDoorState": "Closed", "upTime": 548542 }
I (551.584) ratgdo-http: SSE send to client 10.0.1.108 on channel 2, data: { "lastDoorUpdateAt": 0, "garageDoorState": "Closed", "upTime": 548542 }
```

dkerr64, 2025-03-31:
Yes, I noticed that the version number is wrong.  Not sure why... that is set by the GitHub release workflow.

dkerr64, 2025-04-01:
When ratgdo first starts up with dry contact mode it may not get the door status correct until the door is opened or closed.  However you are seeing a problem after operating the door.  Problem appears here...

First door is detected as closed...
```
D (551.252) gdolib: Wake up for GPIO pin 14 (closed sensor), level is 0
D (551.305) gdolib: Level of pin 14 (closed sensor) is 0. Interrupts during debounce: 2 (0.029ms)
D (551.314) gdolib: Close sensor pressed
D (551.321) gdolib: Door state: Closed
```
And we tell HomeKit that we are closed...
```
I (551.332) ratgdo-homekit: Garage door set CurrentDoorState: 1
```

But oops...  now the sensor says it is no longer closed... (I have deleted some log msgs)
```
D (551.330) gdolib: Wake up for GPIO pin 14 (closed sensor), level is 1
D (551.383) gdolib: Level of pin 14 (closed sensor) is 1. Interrupts during debounce: 0 (0.000ms)
D (551.404) gdolib: Close sensor released
D (551.409) gdolib: Door state: Opening
```
And we tell HomeKit that we are opening....
```
I (551.433) ratgdo-homekit: Garage door set TargetDoorState: 0
I (551.472) ratgdo-homekit: Garage door set CurrentDoorState: 2
```

But now the sensor says closed again...
```
D (551.458) gdolib: Wake up for GPIO pin 14 (closed sensor), level is 0
D (551.522) gdolib: Level of pin 14 (closed sensor) is 0. Interrupts during debounce: 2 (0.028ms)
D (551.529) gdolib: Close sensor pressed
D (551.533) gdolib: Door state: Closed
```
And we go tell HomeKit
```
I (551.543) ratgdo-homekit: Garage door set CurrentDoorState: 1
```

But while we set the CurrentDoorState to 1 (CLOSED) you will note above that TargetDoorState is still 0 (OPEN).  That is wrong, we need to have both current and target set to the same thing.

The flip of the sensor from closed to open to closed is "bouncing".  There is some debounce detection in the code, but it is only set to 50ms.  The bad reading is coming at 551.330, which is 78ms after first triggered at 551.252, so debounce failed.

So... two fixes.

1. Set the debounce to a duration higher than 50ms
2. Make sure that both current and target door states are set to the same thing when open or closed.

I can do the 2nd, and while there is no user interface to change the debounce period, there is a way you can change it...

```
curl -X POST http://<ip-address>/setgdo \
-H "Content-Type: application/x-www-form-urlencoded" \
-d "dcDebounceDuration=250"
```

I picked 250ms but you could go higher or lower.

aberkvam, 2025-04-01:
I tried setting dcDebounceDuration to 250, 500, and 1000.  The behavior didn't change.

Logs for setting it to 1000 and closing the door:
```
I (4892.985) ratgdo-http: Client 10.0.1.108 requesting: /setgdo (method: HTTP_POST)
I (4892.990) ratgdo-http: Configuration set for Key: dcDebounceDuration, Value: 1000
I (4892.995) ratgdo-http: SetGDO Complete
I (4899.137) ratgdo-homekit: Garage Door Characteristics Update
I (4899.143) ratgdo-comms: Delay door close by 2 seconds
I (4899.147) ratgdo-comms: Start function delay timer for 2000ms (8 iterations)
I (4901.413) ratgdo-comms: Calling delayed function 0x40104FA0
I (4901.418) gdolib: Send door close
D (4901.423) gdo_utils: Dry Contact set pin 25 to 1
D (4901.430) gdo_utils: Dry Contact set pin 17 to 1
D (4901.525) gdolib: Wake up for GPIO pin 14 (closed sensor), level is 1
D (4901.526) gdolib: Wake up for GPIO pin 13 (open sensor), level is 0
D (4901.579) gdolib: Level of pin 14 (closed sensor) is 1. Interrupts during debounce: 0 (0.000ms)
D (4901.584) gdolib: Close sensor released
D (4901.589) gdolib: Door state: Open
I (4901.596) ratgdo-comms: GDO event: door: Open,   0%, target: 100%
D (4901.596) gdolib: Level of pin 13 (open sensor) is 0. Interrupts during debounce: 0 (0.000ms)
D (4901.605) gdolib: Open sensor pressed
D (4901.608) gdolib: Door state: Open
D (4901.832) gdolib: Wake up for GPIO pin 13 (open sensor), level is 1
D (4901.885) gdolib: Debounce of pin 13 (open sensor) not stable, wait for another 50ms. Interrupts: 4 (25.956ms)
D (4901.930) gdo_utils: Dry Contact set pin 25 to 0
D (4901.937) gdo_utils: Dry Contact set pin 17 to 0
D (4901.942) gdolib: Level of pin 13 (open sensor) is 0. Interrupts during debounce: 4 (25.956ms)
D (4901.948) gdolib: Open sensor pressed
D (4901.954) gdolib: Door state: Open
D (4902.093) gdolib: Wake up for GPIO pin 13 (open sensor), level is 1
D (4902.147) gdolib: Level of pin 13 (open sensor) is 1. Interrupts during debounce: 0 (0.000ms)
D (4902.153) gdolib: Open sensor released
D (4902.159) gdolib: Door state: Closing
I (4902.166) ratgdo-comms: GDO event: door: Closing,   0%, target: 100%
I (4902.172) ratgdo-http: Current Door State changing from Open to Closing
I (4902.173) ratgdo-homekit: Garage door set TargetDoorState: 1
I (4902.180) ratgdo-http: SSE send to client 10.0.1.108 on channel 0, data: { "lastDoorUpdateAt": 0, "garageDoorState": "Closing", "upTime": 4899174 }
I (4902.200) ratgdo-http: SSE send to client 10.0.1.108 on channel 1, data: { "lastDoorUpdateAt": 0, "garageDoorState": "Closing", "upTime": 4899174 }
I (4902.212) ratgdo-homekit: Garage door set CurrentDoorState: 3
I (4902.216) ratgdo-http: SSE send to client 10.0.1.108 on channel 2, data: { "lastDoorUpdateAt": 0, "garageDoorState": "Closing", "upTime": 4899174 }
I (4902.235) ratgdo-http: SSE send to client 10.0.1.108 on channel 3, data: { "lastDoorUpdateAt": 0, "garageDoorState": "Closing", "upTime": 4899174 }
I (4902.666) ratgdo-comms: GDO event: door: Closing,  13%, target: 100%
I (4903.166) ratgdo-comms: GDO event: door: Closing,  17%, target: 100%
I (4903.666) ratgdo-comms: GDO event: door: Closing,  21%, target: 100%
I (4904.166) ratgdo-comms: GDO event: door: Closing,  26%, target: 100%
I (4904.666) ratgdo-comms: GDO event: door: Closing,  30%, target: 100%
I (4905.166) ratgdo-comms: GDO event: door: Closing,  34%, target: 100%
I (4905.666) ratgdo-comms: GDO event: door: Closing,  39%, target: 100%
I (4906.166) ratgdo-comms: GDO event: door: Closing,  43%, target: 100%
I (4906.666) ratgdo-comms: GDO event: door: Closing,  47%, target: 100%
I (4907.166) ratgdo-comms: GDO event: door: Closing,  52%, target: 100%
I (4907.666) ratgdo-comms: GDO event: door: Closing,  56%, target: 100%
I (4908.167) ratgdo-comms: GDO event: door: Closing,  60%, target: 100%
I (4908.666) ratgdo-comms: GDO event: door: Closing,  65%, target: 100%
I (4909.166) ratgdo-comms: GDO event: door: Closing,  69%, target: 100%
I (4909.666) ratgdo-comms: GDO event: door: Closing,  73%, target: 100%
I (4910.166) ratgdo-comms: GDO event: door: Closing,  78%, target: 100%
I (4910.666) ratgdo-comms: GDO event: door: Closing,  82%, target: 100%
I (4911.166) ratgdo-comms: GDO event: door: Closing,  87%, target: 100%
I (4911.666) ratgdo-comms: GDO event: door: Closing,  91%, target: 100%
I (4912.166) ratgdo-comms: GDO event: door: Closing,  95%, target: 100%
I (4912.666) ratgdo-comms: GDO event: door: Closing, 100%, target: 100%
D (4913.365) gdolib: Wake up for GPIO pin 14 (closed sensor), level is 0
D (4913.423) gdolib: Level of pin 14 (closed sensor) is 0. Interrupts during debounce: 2 (0.028ms)
D (4913.426) gdolib: Close sensor pressed
D (4913.428) gdolib: Door state: Closed
I (4913.431) ratgdo-comms: GDO event: door: Closed, 100%, target: 100%
D (4913.439) gdolib: Wake up for GPIO pin 14 (closed sensor), level is 1
I (4913.448) ratgdo-http: Current Door State changing from Closing to Closed
[... 3 lines cut ...]
I (4913.495) ratgdo-http: SSE send to client 10.0.1.108 on channel 2, data: { "lastDoorUpdateAt": 0, "garageDoorState": "Closed", "upTime": 4910450 }
D (4913.497) gdolib: Level of pin 14 (closed sensor) is 1. Interrupts during debounce: 0 (0.000ms)
I (4913.511) ratgdo-http: SSE send to client 10.0.1.108 on channel 3, data: { "lastDoorUpdateAt": 0, "garageDoorState": "Closed", "upTime": 4910450 }
D (4913.516) gdolib: Close sensor released
D (4913.536) gdolib: Door state: Opening
I (4913.540) ratgdo-comms: GDO event: door: Opening, 100%, target: 100%
[... cut ...]
I (4913.669) ratgdo-http: SSE send to client 10.0.1.108 on channel 0, data: { "lastDoorUpdateAt": 0, "garageDoorState": "Closed", "upTime": 4910663 }
I (4913.689) ratgdo-http: SSE send to client 10.0.1.108 on channel 1, data: { "lastDoorUpdateAt": 0, "garageDoorState": "Closed", "upTime": 4910663 }
I (4913.704) ratgdo-http: SSE send to client 10.0.1.108 on channel 2, data: { "lastDoorUpdateAt": 0, "garageDoorState": "Closed", "upTime": 4910663 }
I (4913.722) ratgdo-http: SSE send to client 10.0.1.108 on channel 3, data: { "lastDoorUpdateAt": 0, "garageDoorState": "Closed", "upTime": 4910663 }
```

dkerr64, 2025-04-01:
Did you reboot between setting debounce duration?  I forgot to mention that.

aberkvam, 2025-04-01:
Ah, that did the trick.  I thought I didn't need to reboot since the logs showed the changed value.  I even thought about asking if I needed to do anything special to save the value so it would persist across reboots.

I rebooted and now I see this in the logs:
```
D (110.405) gdolib: Level of pin 14 (closed sensor) is 0. Interrupts during debounce: 6 (210.925ms)
```

And the HomeKit status of the door seems to be stable and correct.  Thanks!

dkerr64, 2025-04-01:
Thank you.  Next release will have both a fix for target / current state being wrong, and also added a user interface slider for dry contact debounce time... I allow for 50ms to 1000ms range.

dkerr64, 2025-04-01:
closing as (hopefully) all fixed in 3.1.6

closed 2025-04-01 by dkerr64