ratgdo/homekit-ratgdo32 #60: Wall Control Panel Problems

TheGadgetGuy, 2025-04-04:
I have the Chamberlain Wall Control Panel, also known as: 41A7305‑1 and the screen now constantly cycles and then defaults to the following message:

PRESS THE
PUSH BAR
TO ACTIVATE CONTROL
SEE USER MANUAL

When you press the push bar, all it does is cycle and com back to the above message.  Currently you can not open the garage door with the wall control panel now.  Really frustrating.  This all happened after updating to v3.1.6

Any help would be greatly appreciated.

TheGadgetGuy, 2025-04-28:
Here is the recent logs from 3.1.8:

Current server message log...
Server upTime:       0 days 00 hrs 02 mins 31 secs
Server uptime (secs): 151.673
Firmware version: 3.1.8
Free heap: 107996
Minimum heap: 102892

 (87.708) gdolib: cmd=48c (OPENINGS) byte2=ab byte1=75 nibble=0
I (87.709) ratgdo-comms: GDO event: openings: 30123
I (87.720) ratgdo-http: SSE send to client 192.168.1.89 on channel 0, data: { "openingsCount": 30123, "upTime": 84731 }
W (87.828) gdolib: Oversized packet received: 20 bytes, messages pending: 1
I (87.829) gdolib: cmd=308 (PAIRED_DEVICES) byte2=08 byte1=00 nibble=0
W (87.901) gdolib: Oversized packet received: 20 bytes, messages pending: 1
I (87.902) gdolib: cmd=081 (STATUS) byte2=42 byte1=60 nibble=2
W (88.090) gdolib: Oversized packet received: 20 bytes, messages pending: 1
I (88.091) gdolib: cmd=48c (OPENINGS) byte2=ab byte1=75 nibble=0
W (88.164) gdolib: Oversized packet received: 20 bytes, messages pending: 1
I (88.165) gdolib: cmd=308 (PAIRED_DEVICES) byte2=08 byte1=00 nibble=0
W (90.094) gdolib: Oversized packet received: 120 bytes, messages pending: 1
E (90.096) gdolib: RX data signature error: 0x000000
I (90.165) gdolib: Unexpected data; received 60 bytes, processing
W (90.166) gdolib: Oversized packet received: 60 bytes, messages pending: 1
E (90.176) gdolib: RX data signature error: 0x000000
I (90.589) ratgdo-comms: GDO event: motion: Clear
I (90.589) ratgdo-comms: Save rolling code: 10757
I (90.589) ratgdo-homekit: Motion reset
I (90.592) ratgdo-http: SSE send to client 192.168.1.89 on channel 0, data: { "garageMotion": false, "upTime": 87603 }
W (90.717) gdolib: Oversized packet received: 20 bytes, messages pending: 1
I (90.718) gdolib: cmd=081 (STATUS) byte2=42 byte1=60 nibble=2
W (90.908) gdolib: Oversized packet received: 20 bytes, messages pending: 1
I (90.909) gdolib: cmd=48c (OPENINGS) byte2=ab byte1=75 nibble=0
W (90.982) gdolib: Oversized packet received: 20 bytes, messages pending: 1
I (90.983) gdolib: cmd=308 (PAIRED_DEVICES) byte2=08 byte1=00 nibble=0
W (92.481) gdolib: Oversized packet received: 20 bytes, messages pending: 1
I (92.482) gdolib: cmd=080 (GET_STATUS) byte2=01 byte1=00 nibble=0
W (92.570) gdolib: Oversized packet received: 20 bytes, messages pending: 1
I (92.570) gdolib: cmd=081 (STATUS) byte2=42 byte1=60 nibble=2
W (92.759) gdolib: Oversized packet received: 20 bytes, messages pending: 1
I (92.760) gdolib: cmd=48c (OPENINGS) byte2=ab byte1=75 nibble=0
W (92.832) gdolib: Oversized packet received: 20 bytes, messages pending: 1
I (92.833) gdolib: cmd=308 (PAIRED_DEVICES) byte2=08 byte1=00 nibble=0
W (93.040) gdolib: Oversized packet received: 20 bytes, messages pending: 1
I (93.040) gdolib: cmd=080 (GET_STATUS) byte2=01 byte1=00 nibble=0
W (93.523) gdolib: Oversized packet received: 20 bytes, messages pending: 1
I (93.524) gdolib: cmd=080 (GET_STATUS) byte2=01 byte1=00 nibble=0
I (98.374) ratgdo-http: Client 192.168.1.89 requesting: /auth (method: HTTP_GET)
I (154.532) ratgdo-http: Client 192.168.1.89 requesting: /logs.html (HTTP_GET, type: text/html, length: 1497)
I (154.589) ratgdo-http: Client 192.168.1.89 requesting: /logs.js (HTTP_GET, type: text/javascript, length: 2592)
I (154.658) ratgdo-http: Client 192.168.1.89 requesting: /showlog (method: HTTP_GET)
I (154.954) ratgdo-http: Client 192.168.1.89 listening for SSE events on channel 1
W (187.842) gdolib: Oversized packet received: 20 bytes, messages pending: 1
I (187.846) gdolib: cmd=080 (GET_STATUS) byte2=01 byte1=00 nibble=0
W (187.934) gdolib: Oversized packet received: 20 bytes, messages pending: 1
I (187.937) gdolib: cmd=081 (STATUS) byte2=42 byte1=60 nibble=2
W (188.120) gdolib: Oversized packet received: 20 bytes, messages pending: 1
I (188.124) gdolib: cmd=48c (OPENINGS) byte2=ab byte1=75 nibble=0
W (188.193) gdolib: Oversized packet received: 20 bytes, messages pending: 1
I (188.197) gdolib: cmd=308 (PAIRED_DEVICES) byte2=08 byte1=00 nibble=0
W (190.423) gdolib: Oversized packet received: 20 bytes, messages pending: 1
I (190.429) gdolib: cmd=080 (GET_STATUS) byte2=01 byte1=00 nibble=0
I (190.474) ratgdo-main: Free heap dropped to 101184
W (190.518) gdolib: Oversized packet received: 20 bytes, messages pending: 1
I (190.522) gdolib: cmd=081 (STATUS) byte2=42 byte1=60 nibble=2
W (190.705) gdolib: Oversized packet received: 20 bytes, messages pending: 1
I (190.708) gdolib: cmd=48c (OPENINGS) byte2=ab byte1=75 nibble=0
W (190.780) gdolib: Oversized packet received: 20 bytes, messages pending: 1
I (190.783) gdolib: cmd=308 (PAIRED_DEVICES) byte2=08 byte1=00 nibble=0
W (190.984) gdolib: Oversized packet received: 20 bytes, messages pending: 1
[... 264 lines cut ...]
I (360.924) ratgdo-comms: GDO event: door: Closing,  77%, target: 100%
W (361.153) gdolib: Oversized packet received: 20 bytes, messages pending: 1
I (361.157) gdolib: cmd=284 (MOTOR_ON) byte2=00 byte1=00 nibble=0
I (361.424) ratgdo-comms: GDO event: door: Closing,  80%, target: 100%
I (361.924) ratgdo-comms: GDO event: door: Closing,  83%, target: 100%
I (362.423) ratgdo-comms: GDO event: door: Closing,  86%, target: 100%
I (362.924) ratgdo-comms: GDO event: door: Closing,  90%, target: 100%
I (363.424) ratgdo-comms: GDO event: door: Closing,  93%, target: 100%
I (363.924) ratgdo-comms: GDO event: door: Closing,  96%, target: 100%
I (364.424) ratgdo-comms: GDO event: door: Closing,  99%, target: 100%
W (364.741) gdolib: Oversized packet received: 20 bytes, messages pending: 1
[... cut ...]
I (364.932) gdolib: cmd=48c (OPENINGS) byte2=ac byte1=75 nibble=0
I (364.936) ratgdo-comms: GDO event: openings: 30124
I (364.941) ratgdo-http: SSE send to client 192.168.1.89 on channel 0, data: { "openingsCount": 30124, "upTime": 361952 }
I (364.957) ratgdo-http: SSE send to client 192.168.1.89 on channel 1, data: { "openingsCount": 30124, "upTime": 361952 }
W (365.004) gdolib: Oversized packet received: 20 bytes, messages pending: 1
I (365.008) gdolib: cmd=308 (PAIRED_DEVICES) byte2=08 byte1=00 nibble=0``

dkerr64, 2025-04-28:
First thing to try is to select the "use software serial" option in settings (under Door protocol section).  Your log indicates that this is not selected and that _might_ be impacting what you see.

dkerr64, 2025-06-13:
Is this still a problem?  I'd like to close this issue unless we still have a problem.

TheGadgetGuy, 2025-06-13:
No, it is not a problem. Looks like latest version is running stable. I’ll confirm this evening and let you know. No trees were harmed in the sending of this message. However, a large number of electrons were terribly inconvenienced.

dkerr64, 2025-06-13:
Thanks, I just released v3.1.11 so use that to test.

closed 2025-06-13 by dkerr64

robohack, 2025-09-25:
A quick followup:

Since upgrading to v3.3.3 and v2.0.2 I still occasionally briefly see this message on the wall panels, maybe only for the ratgdo32, but it disappears on its own after a very brief appearance and there doesn't seem to be any functional impact -- all the controls and remotes work as expected and all expected notifications are delivered.