ratgdo/homekit-ratgdo32 #61: Feature Request: Expose Lock Remotes to HomeKit

buns-eastern, 2025-04-10:
David - thanks for all your hard work on this project!   Release 3.16 is working perfectly FYI.  Not sure if this has been asked before, if so I apologize.  Can the Lock Remotes function be exposed to HomeKit so it can be set manually via HomeKit or used in automations?  Thanks in advance!

dkerr64, 2025-04-10:
The lock accessory is exposed to HomeKit.  The Apple Home app seems to hide that (but it does give notifications when it locks/unlocks).  But the Eve app does show the lock it and you can control it from there.  And if you create any scenes/automations in Eve then they will be visible in Apple Home app.

john-ke5c, 2025-04-14:
Have you tried:

Home > Garage door > Accessory settings > "show as separate tiles"?

P.S. I am using Ratgdo-Homebridge which I know is different, but this may be a Homekit settings issue.

buns-eastern, 2025-04-24:
Thanks again for the tip about using the Eve app...works really well!  I would have never figured that out!

dkerr64, 2025-04-26:
Closing as no work required.

closed 2025-04-26 by dkerr64