ratgdo/homekit-ratgdo32 #63: Still having issues with builtin TTC on 3.1.x

jw7519, 2025-04-12:
Hello,

I think I spoke too soon about issue https://github.com/ratgdo/homekit-ratgdo32/issues/50 being entirely fixed.

Turns out that the builtin TTC now seems to be stuck at 5 mins.  I'm not really sure why.  I haven't messed with settings, but I can't keep my door open for longer than 5 mins at a time. 

I have tho Door Close Delay currently set to 5 seconds.  I've enabled and then disabled the TTC and I can see in the logs that the value 0 is being set, but that seems to have no effect.  The door still closes after 5 mins.

Any ideas on how to resolve?

Here are the logs:

```logs
Current server message log...
Server upTime:       20 days 08 hrs 43 mins 43 secs
Server uptime (secs): 1759423.392
Firmware version: 3.1.3
Free heap: 76640
Minimum heap: 76640

teAt": 0, "garageDoorState": "Closed", "upTime": 1759202703 }
E (1759205.758) gdolib: Collision detected trying to send packet
I (1759206.181) gdolib: cmd=48c (OPENINGS) byte2=b8 byte1=00 nibble=0
I (1759206.182) gdolib: cmd=308 (PAIRED_DEVICES) byte2=05 byte1=00 nibble=0
I (1759206.193) ratgdo-comms: GDO event: openings: 184
I (1759206.193) ratgdo-http: SSE send to client 10.10.4.119 on channel 0, data: { "openingsCount": 184, "upTime": 1759203188 }
I (1759347.408) ratgdo-http: Client 10.10.4.119 requesting: /auth (method: HTTP_GET)
I (1759365.827) ratgdo-http: Client 10.10.4.119 requesting: /auth (method: HTTP_GET)
I (1759391.302) ratgdo-http: Client 10.10.4.119 requesting: /auth (method: HTTP_GET)
I (1759410.403) ratgdo-http: Client 10.10.4.119 requesting: /index.html (HTTP_GET, type: text/html, length: 6460)
I (1759410.460) ratgdo-http: Client 10.10.4.119 requesting: /status.json (method: HTTP_GET)
I (1759410.578) ratgdo-http: JSON length: 1233
I (1759411.432) ratgdo-http: Sending 404 Not Found for: /logs with method: HTTP_GET to client: 10.10.4.119
I (1759411.477) ratgdo-http: Sending 404 Not Found for: /favicon.ico with method: HTTP_GET to client: 10.10.4.119
I (1759411.725) ratgdo-http: Client 10.10.4.119 not listening, remove SSE subscription. Total subscribed: 0
I (1759414.248) ratgdo-http: Client 10.10.4.119 requesting: /index.html (HTTP_GET, type: text/html, length: 6460)
I (1759414.310) ratgdo-http: Client 10.10.4.119 requesting: /status.json (method: HTTP_GET)
I (1759414.427) ratgdo-http: JSON length: 1233
I (1759414.466) ratgdo-http: Client 10.10.4.119 requesting: /rest/events/subscribe (method: HTTP_GET)
I (1759414.468) ratgdo-http: SSE Subscription for client b4ac541c-7420-4c1d-bfe8-ea0fda18f635 with IP 10.10.4.119: event bus location: /rest/events/0, Total subscribed: 1
I (1759414.528) ratgdo-http: Client 10.10.4.119 listening for SSE events on channel 0
I (1759425.310) ratgdo-http: Client 10.10.4.119 requesting: /logs.html (HTTP_GET, type: text/html, length: 1498)
I (1759425.339) ratgdo-http: Client 10.10.4.119 requesting: /logs.js (HTTP_GET, type: text/javascript, length: 2592)
I (1759425.398) ratgdo-http: Client 10.10.4.119 requesting: /status.json (method: HTTP_GET)
I (1759425.515) ratgdo-http: JSON length: 1233
I (1759425.524) ratgdo-http: Client 10.10.4.119 requesting: /showrebootlog (method: HTTP_GET)
I (1759425.530) ratgdo-logger: Print saved log
I (1759425.549) ratgdo-http: Client 10.10.4.119 requesting: /crashlog (method: HTTP_GET)
I (1759425.551) ratgdo-http: Request to display crash log...
I (1759425.657) ratgdo-main: Free heap dropped to 76640
I (1759425.666) ratgdo-http: Client 10.10.4.119 requesting: /rest/events/subscribe (method: HTTP_GET)
I (1759425.670) ratgdo-http: SSE Subscription for client c7da9995-4247-447c-89b7-21a88ab48225 with IP 10.10.4.119: event bus location: /rest/events/1, Total subscribed: 2
I (1759425.727) ratgdo-http: Client 10.10.4.119 listening for SSE events on channel 1
I (1759426.390) ratgdo-http: Client 10.10.4.119 requesting: /showlog (method: HTTP_GET)
I (1759425.727) ratgdo-http: Client 10.10.4.119 listening for SSE events on channel 1
I (1759426.390) ratgdo-http: Client 10.10.4.119 requesting: /showlog (method: HTTP_GET)
I (1759446.488) ratgdo-http: Client 10.10.4.119 requesting: /status.json (method: HTTP_GET)
I (1759446.608) ratgdo-http: JSON length: 1233
I (1759457.565) ratgdo-http: Client 10.10.4.119 requesting: /status.json (method: HTTP_GET)
I (1759457.686) ratgdo-http: JSON length: 1233
I (1759460.176) ratgdo-http: Client 10.10.4.119 requesting: /auth (method: HTTP_GET)
I (1759505.684) gdolib: cmd=081 (STATUS) byte2=00 byte1=60 nibble=2
I (1759505.710) ratgdo-comms: GDO event: light: Off
I (1759505.713) ratgdo-comms: Save rolling code: 60337
I (1759505.720) ratgdo-http: SSE send to client 10.10.4.119 on channel 0, data: { "garageLightOn": false, "upTime": 1759502714 }
I (1759505.715) ratgdo-homekit: Light has turned off
I (1759505.737) ratgdo-http: SSE send to client 10.10.4.119 on channel 1, data: { "garageLightOn": false, "upTime": 1759502714 }
I (1759505.934) gdolib: cmd=48c (OPENINGS) byte2=b8 byte1=00 nibble=0
I (1759506.182) gdolib: cmd=308 (PAIRED_DEVICES) byte2=05 byte1=00 nibble=0
I (1759533.474) ratgdo-http: Client 10.10.4.119 requesting: /status.json (method: HTTP_GET)
I (1759533.595) ratgdo-http: JSON length: 1234
I (1759542.341) ratgdo-http: Client 10.10.4.119 requesting: /auth (method: HTTP_GET)
I (1759542.436) ratgdo-http: Client 10.10.4.119 requesting: /setgdo (method: HTTP_POST)
I (1759542.441) ratgdo-http: Configuration set for Key: builtInTTC, Value: 1
I (1759542.448) ratgdo-http: SetGDO Complete
I (1759542.499) ratgdo-http: Client 10.10.4.119 requesting: /index.html (HTTP_GET, type: text/html, length: 6460)
I (1759542.528) ratgdo-http: Client 10.10.4.119 not listening, remove SSE subscription. Total subscribed: 1
I (1759542.586) ratgdo-http: Client 10.10.4.119 requesting: /status.json (method: HTTP_GET)
I (1759542.707) ratgdo-http: JSON length: 1233
I (1759542.744) ratgdo-http: Client 10.10.4.119 requesting: /rest/events/subscribe (method: HTTP_GET)
I (1759542.750) ratgdo-http: SSE Subscription for client fcc4904b-c053-47c2-bec9-9fcb8f8758e3 with IP 10.10.4.119: event bus location: /rest/events/0, Total subscribed: 2
I (1759542.810) ratgdo-http: Client 10.10.4.119 listening for SSE events on channel 0
I (1759585.144) ratgdo-http: Client 10.10.4.119 requesting: /status.json (method: HTTP_GET)
I (1759585.266) ratgdo-http: JSON length: 1233
I (1759587.405) ratgdo-http: Client 10.10.4.119 requesting: /auth (method: HTTP_GET)
I (1759605.718) ratgdo-http: Client 10.10.4.119 requesting: /auth (method: HTTP_GET)
I (1759605.804) ratgdo-http: Client 10.10.4.119 requesting: /setgdo (method: HTTP_POST)
I (1759605.810) ratgdo-http: Configuration set for Key: builtInTTC, Value: 0
I (1759605.818) ratgdo-config: Disable built-in TTC, set to: 0
I (1759605.822) gdolib: TTC Seconds remaining: 0
I (1759605.848) ratgdo-comms: GDO event: set TTC: 0
I (1759605.851) ratgdo-http: SetGDO Complete
I (1759605.887) ratgdo-http: Client 10.10.4.119 requesting: /index.html (HTTP_GET, type: text/html, length: 6460)
I (1759605.959) ratgdo-http: Client 10.10.4.119 requesting: /status.json (method: HTTP_GET)
I (1759606.081) ratgdo-http: JSON length: 1234
I (1759606.162) ratgdo-http: Client 10.10.4.119 requesting: /rest/events/subscribe (method: HTTP_GET)
I (1759606.167) ratgdo-http: SSE Subscription for client f9c20aa5-784f-4f87-9afb-11bef7af3af4 with IP 10.10.4.119: event bus location: /rest/events/2, Total subscribed: 3
I (1759606.293) ratgdo-http: Client 10.10.4.119 listening for SSE events on channel 2
I (1759606.811) ratgdo-http: Client 10.10.4.119 not listening, remove SSE subscription. Total subscribed: 2
I (1759729.043) ratgdo-http: Client 10.10.4.119 requesting: /status.json (method: HTTP_GET)
I (1759729.164) ratgdo-http: JSON length: 1234
I (1759731.110) ratgdo-http: Client 10.10.4.119 requesting: /auth (method: HTTP_GET)
I (1759805.684) gdolib: cmd=081 (STATUS) byte2=00 byte1=60 nibble=2
I (1759805.933) gdolib: cmd=48c (OPENINGS) byte2=b8 byte1=00 nibble=0
I (1759806.182) gdolib: cmd=308 (PAIRED_DEVICES) byte2=05 byte1=00 nibble=0
I (1759861.747) ratgdo-http: Client 10.10.4.119 requesting: /status.json (method: HTTP_GET)
I (1759861.869) ratgdo-http: JSON length: 1234
I (1759919.716) ratgdo-http: Client 10.10.4.119 requesting: /status.json (method: HTTP_GET)
I (1759919.838) ratgdo-http: JSON length: 1234
```

dkerr64, 2025-04-12:
The logs show that we are setting TTC to zero any the garage door...

```
I (1759605.810) ratgdo-http: Configuration set for Key: builtInTTC, Value: 0
I (1759605.818) ratgdo-config: Disable built-in TTC, set to: 0
I (1759605.822) gdolib: TTC Seconds remaining: 0
I (1759605.848) ratgdo-comms: GDO event: set TTC: 0
I (1759605.851) ratgdo-http: SetGDO Complete
```

So I am not sure why your door continues to close after 5 minutes.

@descipher can you advise what else we could try?

@jw7519 I notice you are on version 3.1.3,  the current version is 3.1.6.  However, if you update to that then you will lose ability to toggle on/off the built-in TTC from the user interface.  There are no changes between the versions that would impact this problem so I suggest you stay on 3.1.3 until we nail this down.

descipher, 2025-04-13:
@dkerr64 We need to find out what security+ is doing.

e.g. 
```
[17:49:43][D][number:054]: 'Time Till Close' - Setting number value
[17:49:43][D][number:113]:   New number value: 300.000000
[17:49:43][D][esp-idf:000]: I (1123722494) gdolib: TTC Seconds remaining: 300
[17:49:43][D][number:012]: 'Time Till Close': Sending state 300.000000
[17:49:43][I][secplus_gdo:126][gdo_main_task]: Set Time to close: 300

[17:49:43][D][esp-idf:000][gdo_main_task]: I (1123722648) gdolib: received rolling=0092af9 fixed=547a0b3006 data=2c010101
[17:49:43][D][esp-idf:000][gdo_main_task]: I (1123722652) gdolib: cmd=401 (PAIR_2_RESP) byte2=2c byte1=01 nibble=1
[17:49:43][D][esp-idf:000][gdo_main_task]: I (1123722721) gdolib: received rolling=0092afa fixed=507a0b3006 data=000c01a1
[17:49:43][D][esp-idf:000][gdo_main_task]: I (1123722724) gdolib: cmd=0a1 (PAIR_3_RESP) byte2=00 byte1=0c nibble=1

[17:51:24][D][number:054]: 'Time Till Close' - Setting number value
[17:51:25][D][number:113]:   New number value: 240.000000
[17:51:25][D][esp-idf:000]: I (1123823853) gdolib: TTC Seconds remaining: 240
[17:51:25][D][number:012]: 'Time Till Close': Sending state 240.000000
[17:51:25][I][secplus_gdo:126][gdo_main_task]: Set Time to close: 240
[17:51:25][D][esp-idf:000][gdo_main_task]: I (1123824007) gdolib: received rolling=0092afd fixed=547a0b3006 data=f0000101
[17:51:25][D][esp-idf:000][gdo_main_task]: I (1123824010) gdolib: cmd=401 (PAIR_2_RESP) byte2=f0 byte1=00 nibble=1
[17:51:25][D][esp-idf:000][gdo_main_task]: I (1123824083) gdolib: received rolling=0092afe fixed=507a0b3006 data=000c01a1
[17:51:25][D][esp-idf:000][gdo_main_task]: I (1123824088) gdolib: cmd=0a1 (PAIR_3_RESP) byte2=00 byte1=0c nibble=1

[17:51:31][D][number:054]: 'Time Till Close' - Setting number value
[17:51:31][D][number:113]:   New number value: 180.000000
[17:51:31][D][esp-idf:000]: I (1123830745) gdolib: TTC Seconds remaining: 180
[17:51:31][D][number:012]: 'Time Till Close': Sending state 180.000000
[17:51:31][I][secplus_gdo:126][gdo_main_task]: Set Time to close: 180
[17:51:32][D][esp-idf:000][gdo_main_task]: I (1123830900) gdolib: received rolling=0092aff fixed=547a0b3006 data=b4000101
[17:51:32][D][esp-idf:000][gdo_main_task]: I (1123830903) gdolib: cmd=401 (PAIR_2_RESP) byte2=b4 byte1=00 nibble=1
[17:51:32][D][esp-idf:000][gdo_main_task]: I (1123830973) gdolib: received rolling=0092b00 fixed=507a0b3006 data=000c01a1
[17:51:32][D][esp-idf:000][gdo_main_task]: I (1123830977) gdolib: cmd=0a1 (PAIR_3_RESP) byte2=00 byte1=0c nibble=1

[17:51:37][D][number:054]: 'Time Till Close' - Setting number value
[17:51:37][D][number:113]:   New number value: 120.000000
[17:51:37][D][esp-idf:000]: I (1123836384) gdolib: TTC Seconds remaining: 120
[17:51:37][D][number:012]: 'Time Till Close': Sending state 120.000000
[17:51:37][I][secplus_gdo:126][gdo_main_task]: Set Time to close: 120
[1[17:51:37][D][esp-idf:000][gdo_main_task]: I (1123836536) gdolib: received rolling=0092b01 fixed=547a0b3006 data=78000101
[17:51:37][D][esp-idf:000][gdo_main_task]: I (1123836539) gdolib: cmd=401 (PAIR_2_RESP) byte2=78 byte1=00 nibble=1
[17:51:37][D][esp-idf:000][gdo_main_task]: I (1123836609) gdolib: received rolling=0092b02 fixed=507a0b3006 data=000c01a1
[17:51:37][D][esp-idf:000][gdo_main_task]: I (1123836613) gdolib: cmd=0a1 (PAIR_3_RESP) byte2=00 byte1=0c nibble=1

[17:51:41][D][number:054]: 'Time Till Close' - Setting number value
[17:51:41][D][number:113]:   New number value: 60.000000
[17:51:41][D][esp-idf:000]: I (1123840142) gdolib: TTC Seconds remaining: 60
[17:51:41][D][number:012]: 'Time Till Close': Sending state 60.000000
[17:51:41][I][secplus_gdo:126][gdo_main_task]: Set Time to close: 60
[17:51:41][D][esp-idf:000][gdo_main_task]: I (1123840299) gdolib: received rolling=0092b03 fixed=547a0b3006 data=3c000101
[17:51:41][D][esp-idf:000][gdo_main_task]: I (1123840303) gdolib: cmd=401 (PAIR_2_RESP) byte2=3c byte1=00 nibble=1
[17:51:41][D][esp-idf:000][gdo_main_task]: I (1123840374) gdolib: received rolling=0092b04 fixed=507a0b3006 data=000c01a1
[17:51:41][D][esp-idf:000][gdo_main_task]: I (1123840377) gdolib: cmd=0a1 (PAIR_3_RESP) byte2=00 byte1=0c nibble=1

[17:51:44][D][number:054]: 'Time Till Close' - Setting number value
[17:51:44][D][number:113]:   New number value: 0.000000
[17:51:44][D][esp-idf:000]: I (1123842890) gdolib: TTC Seconds remaining: 0
[17:51:44][D][number:012]: 'Time Till Close': Sending state 0.000000
[17:51:44][I][secplus_gdo:126][gdo_main_task]: Set Time to close: 0
[17:51:44][D][esp-idf:000][gdo_main_task]: I (1123843022) gdolib: received rolling=0092b05 fixed=507a0b3006 data=000901a1
[17:51:44][D][esp-idf:000][gdo_main_task]: I (1123843025) gdolib: cmd=0a1 (PAIR_3_RESP) byte2=00 byte1=09 nibble=1
```

I have confirmed it does work correctly using SecplusGDO and it has this set of responses.
Setting starts at 300s decrementing to 0s
Make sure the data results are correct. PAIR_2_RESP have the set value byte2 = 2C  byte1 = 01  or hex 12C = 300s
PAIR_3_RESP appears to enable/disable of the function with  byte1 = 00 byte2 = 0c (enabled)

the count down shows PAIR_2_RESP only changes the time value. But at 0s PAIR_3_RESP changes to byte2 = 09 (disabled or hold on the wall panel) that should be an indication it working correctly.

jw7519, 2025-05-10:
@dkerr64 This issue is still affecting me.  Is there anything I can try to help debug?

dkerr64, 2025-05-10:
On the settings page please disable the flash lights during TTC and see if that makes a difference.

jw7519, 2025-05-11:
Sorry I don't see that setting?

[image]

dkerr64, 2025-05-11:
You need to upgrade to latest firmware, v3.1.8... which should have solved your original problem without turning off the light flashing.

[image]

jw7519, 2025-05-12:
@dkerr64 I'm a bit confused.  I thought you asked me not to update because the issue was specifically with the built-in TTC and newer versions did not change those settings. 

Has this changed?   Does 3.1.8 resolve?

dkerr64, 2025-05-13:
@jw7519 Sorry, I was getting your issue mixed up with a different problem.  The later versions do not have the option to adjust the built-in TTC (at least not in the web page).

So, to confirm, the current status is that your door is automatically closing after 5 minutes... and setting the TTC to zero is not changing that?  Correct?

@descipher any suggestions on how to turn off the built-in TTC given above?

Thanks

GelidusResearch, 2025-05-13:
Yes, there is a GDO reset full reset function available, it will clear all remote clients and all optional sets. Press and hold the yellow button until the status LED flashes 6 times. Then reboot the RATGDO and re-pair all wireless remote clients.

dkerr64, 2025-06-08:
@jw7519 were you able to reset your door such that it is not automatically closing?

dkerr64, 2025-06-09:
Make sure you do the reset three times.  Google AI states "To factory reset a LiftMaster garage door opener, locate the "Learn" button, usually yellow or purple, on the back of the opener motor. Press and hold the button until the LED light next to it turns off (about 6 seconds). Repeat this process for a second and third time, and you should hear three consecutive beeps, indicating a successful reset."

https://www.wikihow.com/Reset-a-Liftmaster-Garage-Door-Opener

dkerr64, 2025-09-13:
Please test with pre-release v3.3.1 (select pre-release box on update dialog).  If you are unable to update from GitHub please download firmware from [here](https://github.com/ratgdo/homekit-ratgdo32/releases/download/v3.3.1/homekit-ratgdo32-v3.3.1.firmware.bin) and install from computer.

Closing this issue, please open a new issue if problems found.

closed 2025-09-13 by dkerr64