ratgdo/homekit-ratgdo32 #64: Security = 1.0 & change firmware?

john-ke5c, 2025-04-14:
1) Would you expect this firmware to work with an old Sears (Chamberlin purple learn button) Security+ 1.0 wall panel and opener?

2) If so, can I just load this firmware over ESPHome using the Ratgdo web UI?

Brief background: I using ratgdo-homebridge because I have a homebridge (and a Hubitat C8 Pro) hub(s). I have the homebridge hub because the plugin/driver for my alarm system is only fully functional there. I didn't try this direct Homekit firmware since I wasn't sure it would work with Security+ 1.0, but I am having some issues with the homebridge version (motion not detected, light button on panel detected but light not coming on, etc.), so now I would like to try it if possible. I just noticed I can't get into the ratgdo web UI, but I haven't rebooted everything yet. TIA!

dkerr64, 2025-04-14:
If ESPhome firmware is working for you, then the native HomeKit firmware should also work.

However, you cannot simply upload one firmware over the other using the webpage, they are different enough that this will not work.  Instead you will have to plug the ratgdo32 board into USB port and install the firmware using Chrome browser and instructions in the readme... be sure to "erase" during that install process.

I cannot say whether it will work properly... my own Sec+1.0 door does not have a motion detector in the wall panel.  It does have a light button and that works fine, as does turning the light on/off from HomeKit.

john-ke5c, 2025-04-15:
Thank you. ESPhome homebridge-firmware was only sort of working: 1) light switch on wall remote had no effect on opener light; 2) motion was not detected; 3) the lock button toggled the lock state, but the green led on the wall unit flashed for only a few seconds instead of continuously when the state was active. 

This firmware does provide a working light switch, but like the ESPhome version does not provide motion detection and the lock button acts the same way providing a flashing led for only several seconds unlike the native opener where the led flashed so long as the remotes were locked out.

The ESPhome firmware with the homebridge app was better with Homekit: 1) it provided an accessory lock switch to allow locking and unlocking the remotes. That switch therefore also told the user the current lock state. As [@buns-eastern](https://github.com/buns-eastern) reported recently, I can find no accessory lock state controlling switch in Homekit. What I can do, however, is use the garage opener lock state to trigger an automation, e.g., "when garage door locks, turn garage light on, then turn off after 1 minute." I know that is kind of useless, but it demonstrates the lock state change can trigger an automation.

My primary request is for the remote lock state to be not only a status but also a switchable action in Homekit, to have a switch as is done in ratgdo-homebridge.

The motion detection I can work around by simply adding an independent motion sensor to my garage, but it does not seem to be reported to Homekit just as the homebridge firmware also fails to do.

Thank you for you hard work to produce this software!

closed 2025-04-26 by dkerr64