ratgdo/homekit-ratgdo32 #65: Obstruction Trigger for Motion

bschuchert, 2025-04-15:
Hello. Thanks for all of the work you have done thus far. This setup works really well and was pretty easy to use out of the box. The only issue I have noticed is with regard to obstruction sensor and how it can be used. I may be misunderstanding how this is supposed to work but I have been unsuccessful getting the obstruction sensor to trigger a motion event. My goal is to use that event to then trigger the light to turn on but for now, I am just trying to see the motion label flip to True by blocking the obstruction sensor*. 

My control pad has a motion detector on it and immediately reports when it detects motion. For the "Motion Triggers" setting, I have the default motion sensor checked (gray state) and I have checked obstruction, but when I step in front of the obstruction sensor, I dont see garageObstructed flip to true and also dont see the same for garageMotion. 

Reagrding the wiring of the obstruction sensors, they appear to be wired in correctly as they will immediately stop the door and reverse it when I test it during a door close scenario. When I try flipping the wires, they no longer function so I think they are properly wired. 

Before we consider this an issue, am I correct to assume the expected behavior of adding obstruction sensor as an option in the Motion Triggers setting is to show motion when someone passes through the obstruction sensor? 

Thanks!

dkerr64, 2025-04-26:
Acknowledging that something is not right here.  I am looking at it and will publish a fix as soon as it is ready.  Thanks for opening the issue.

dkerr64, 2025-04-27:
believe fixed in 3.1.7

closed 2025-04-27 by dkerr64

bschuchert, 2025-04-28:
I updated this morning to 3.1.8 as that was the latest. From my testing today, this is still an issue. I do not see the control panel shift to "motion: true" or "obstruction: true" when I block the obstruction sensor with my foot. When blocking the obstruction sensor, I should see both of those go true, correct?

Quick note that may or may not be related:
I encountered an issue where the panel was "locked" and would not allow a remote to work. This appears to have happened post firmware update. I overrode this by toggling the lock option off on the panel. 

Is there a log level and event I can inspect to help you debug this?