ratgdo/homekit-ratgdo32 #67: New install issue

chdmlr, 2025-04-21:
Trying to install a ratgdo32 via Homekit.

When I initially booted up the chip it was flashing blue, and I connected to the Ratgdo wifi network and tried installing ESP firmware while following a youtube video before realizing that wasn't what I wanted/needed. Reset the device via the Console & Logs page before doing the steps below. 

1. Plugged in via USB to macbook pro
2. Navigated to [Flash installer](https://ratgdo.github.io/homekit-ratgdo32/flash.html) in Chrome,
3. Click 'connect' via USB single serial
4. Clicked install homekit-ratgdo, confirmed install homekit-ratgdo v3.1.6
5. After a few "failed to initialize" errors and timing out once, it did go through the installation process and gave me the confirmation that install was complete

Then nothing happened. Just back to Flash installer page with the Connect button. No prompt to configure an SSID. No Garage Door XXXXX wifi network popping up in my networks. When I click on connect again and try and follow the same steps it just gives me the failed to initialize logo. 

Power cycled the device a few times and got it to go through install again with "erase device" checked and it got through to the confirmation and again, nothing else.

Any help on what I'm doing wrong or am I missing a step?

dkerr64, 2025-04-26:
If you are not getting the prompt to configure WiFi as part of the install process, then try using the WiFi network created by the ratgdo, and then pointing your browser to 192.168.4.1.  See documentation [here](https://github.com/ratgdo/homekit-ratgdo32?tab=readme-ov-file#how-do-i-install-it).

dkerr64, 2025-06-08:
@chdmlr were you able to install okay?

dkerr64, 2025-06-13:
Closing this issue as no response.

closed 2025-06-13 by dkerr64