ratgdo/homekit-ratgdo32 #70: Cannot get ratgdo32 disco working with 8550WLB & HomeKit 3.1.6 firmware

ryanburnett, 2025-04-27:
I just received my ratgdo32 disco and cannot get it working with the HomeKit 3.1.6 firmware. Here's my setup:

- ratgdo32 disco x2
- HomeKit 3.1.6 firmware
- LiftMaster 8550WLB Security + 2.0 garage door opener x2
- M4 Pro Macbook Pro running macOS 15.4.1
- Chrome Version 135.0.7049.115

Here are my steps to reproduce:

- Bring one disco and my MacBook Pro into my garage
- Connect the disco to my MacBook Pro with a USB-C cable
- Open Chrome and go to:
- https://ratgdo.github.io/homekit-ratgdo32/flash.html
- Click Connect
- Click USB Single Serial
- Click Connect
- Click Install HomeKit-ratgdo
- Check Erase device
- Click Next, Install, Next
- Enter the network password
- Click Connect
- Click Visit Device
- View the disco web UI at its IP address (http://10.1.1.239/)
- Scan the HomeKit QR code to add the disco to the iOS Home app
- Disconnect the disco from my MacBook Pro
- Connect the JST Harness to the disco and its red, white, and black wires to the 8550WLB red, white (next to red), and black terminals
- Connect the wall opener red and white wires to the disco R and W terminals
- Connect the sensor white and black wires to the disco W and B terminals
- Connect the disco USB power supply to an outlet and the USB power cable to the disco
- WiFi Signal is -49 dBm
- The disco intermittently beeps roughly every 20 seconds
- I can no longer access the disco web UI at its IP address
- I can open the garage door with the Home app, but I cannot close the garage door
- After I use the Home app or wall opener to try to close the garage door, the wall opener display says "Sensor momentarily obstructed, error code 4-6, see user manual"
- The disco keeps intermittently beeping roughly every 20 seconds

When I install the 2025.3.3 ESPHome firmware and use it with Home Assistant, everything works great.

Anything else I can try to get the disco working with the HomeKit firmware? Thank you.

dkerr64, 2025-04-27:
Acknowledging receipt of this report.  Not sure at the moment what could cause this.  Tagging @descipher for his awareness as may be related to GDOLIB.

GelidusResearch, 2025-04-27:
Sound like its sensing an obstruction for a brief moment and the firmware of this opener handles it differently that previously tested. The main difference in RATGDO vs GDOLIB code is this 700ms delay before declaring an obstruction:
~~~
                } else {
                    // if the line is high and was last asleep more than 700ms ago, then there is an obstruction present
                    if (current_millis - last_asleep > 700) {
                        this->obstruction_state = ObstructionState::OBSTRUCTED;
                    }
                }
 ~~~

The issue with a nearly 1s wait is that the detection is retarded some what so we do not have a fast indication of an actual fault.
This can be an issue if there is any thread blocking and maybe that what we are observing. But in this case do we assume the sensor is the one that is sleeping? If it is then we need to wait for it and that's not present in the GDOLIB, it does not wait it counts the pulses via IRQ processing and if the pulse does not occur it alarms on it via the event loop.

dkerr64, 2025-04-27:
I'm not sure that this is related to obstruction sensing.  I am concerned that the ratgdo is beeping every 20 seconds, that is bad.  About the only time the HomeKit firmware beeps is just after a reboot, to signal all is initialized... if it is doing that every 20 seconds then it is crashing and rebooting in a loop.

We should be able to get the traceback for the crash if you can boot the ratgdo to a stable state and get to the web page.  You may have to disconnect it from the GDO to avoid the crash.  But the bottom of the web page should indicate a crash count, if not zero then you can get to the crash log and retrieve the traceback.

dkerr64, 2025-04-27:
I have just releases 3.1.7 which has the change that @GelidusResearch suggested... though I am skeptical that it will help.  When the ratgdo boots you will get a beep.  If it continues to beep then it is in a crash/reboot loop and we need to figure out why.  Either monitor the serial (USB) port log, or get the traceback from the web page.

ryanburnett, 2025-04-27:
It's working perfectly now!

Here's what I did with my first disco:

- Disconnected the disco from the GDO
- Connected it to my MacBook Pro with a USB-C cable
- Opened Chrome and went to https://ratgdo.github.io/homekit-ratgdo32/flash.html
- Clicked Connect, selected USB Single Serial, Connect, and clicked Install HomeKit-ratdgo
- Checked Erase device and clicked Next
- "Do you want to install homekit-ratgdo v3.1.7?" > Install
- The disco beeped once, the screen said "Installation complete!" and I clicked Next
- Selected my network SSID, entered the password, and clicked connect
- The disco beeped one tone, then a different tone a second later, then the same tone seven seconds later
- The screen said "Device connected to the network!" and I clicked Visit Device
- Chrome says "This site can’t be reached, http://10.1.1.242/ is unreachable, ERR_ADDRESS_UNREACHABLE" and had a "Reload" button, which had the same result (this also happened with 3.1.6)
- However, I can open http://10.1.1.242 in Safari and view the disco web UI showing "Firmware: v3.1.7"
- I disconnected the disco and connected it to the USB power cable by my GDO
- It beeped right when I plugged it in, then beeped a different tone seven seconds later - no more beeps after that
- I can still connect to the web UI and it shows "Show system logs (crashCount: 0)"
- Enabled the laser in settings
- Added the bridge, door, laser, and light to the Home app
- Hooked up the seven wires to/from the disco
- Opened the GDO in the Home app successfully
- Closed the GDO in the Home app successfully

Repeated the above steps on my second disco with the same successful results.

This is awesome, I can delete my 32GB VMware Fusion Home Assistant VM now and just use HomeKit.

Thanks for all your hard work!

dkerr64, 2025-04-27:
Thanks for reporting back.  Bit by bit we are making this firmware more stable and useful.

closed 2025-04-27 by dkerr64