ratgdo/homekit-ratgdo32 #72: ratgdo32 discos offline less than 24 hours after installing HomeKit 3.1.7

ryanburnett, 2025-04-28:
As a follow up to yesterday's #70, less than 24 hours later both of my ratgdo32 discos:

- Are no longer reachable by their IP address (web UI or ping)
- No longer work in the Home app ("No Response")
- Start intermittently beeping after I use the wall opener to open their respective garage door

Video:

- Left garage door (after I opened it): https://youtu.be/Ctq0l_zTSf4
- Right garage door (before I opened it, you can also hear the other disco beeping in this video): https://youtu.be/ZN1WTLfawBc

Setup:

- ratgdo32 disco x2
- HomeKit 3.1.7 firmware
- LiftMaster 8550WLB Security + 2.0 garage door opener x2
- M4 Pro Macbook Pro running macOS 15.4.1
- Chrome Version 135.0.7049.115

Please let me know what I should try next.

Thank you!

dkerr64, 2025-04-28:
I'm pretty sure that you are in a boot/crash/boot/crash loop.  During a normal boot-up there are two beeps of different tone.  The first beep happens immediately on [startup](https://github.com/ratgdo/homekit-ratgdo32/blob/0fde9b7378a2acb9767993a8e4175353375aca2b/src/ratgdo.cpp#L97) and the second beep with a higher pitched tone on [successful startup](https://github.com/ratgdo/homekit-ratgdo32/blob/0fde9b7378a2acb9767993a8e4175353375aca2b/src/homekit.cpp#L110).  The two beeps are typically about 10 seconds apart (depending on how quickly WiFi connects). We are not getting that far, I am hearing the same very first beep repeat.

The question is how far is it getting.  If you have a way to monitor the serial port then you could capture the logs.  This is very much a developer skill though (I have a long USB cable running from my garage door opener into my basement) but has the advantage of being able to capture the crash traceback when it occurs.

Slightly easier would be to setup a syslog server.  As soon as WiFi is established, we can start logging to a syslog server and see how far it gets (but would first need to get it to boot once to set that up... but I think you said that you can do that if disconnected from the door).  This would  narrow it down as we can see what the last log message was, but the crash traceback will not be captured.

ryanburnett, 2025-04-29:
I setup a syslog server on my PC and:

- Connected the disco to my MacBook Pro
- Installed 3.1.8
- Connected it to the Wi-Fi
- Connected it to the syslog server
- Rebooted the disco
- Confirmed it was sending syslog messages to the server
- Disconnected the disco from my MacBook Pro
- Connected the disco to its USB power supply by my GDO
- Heard the two different beep tones
- Didn't hear any additional beeps
- Connected the wires to the GDO and disco
- Checked my PC and noticed the last message was from about two minutes ago
- Here's the log:

```
<134>1 - Garage-Door-F26E04 ratgdo-homekit 0 - - WiFi established, count: 1, IP: 10.1.1.242, Mask: 255.255.255.0, Gateway: 10.1.1.1, DNS: 10.1.1.40
<134>1 - Garage-Door-F26E04 ratgdo-vehicle 0 - - === Setup VL53L4CX time-of-flight sensor ===
<134>1 - Garage-Door-F26E04 ratgdo-main 0 - - Free heap dropped to 132268
<134>1 - Garage-Door-F26E04 gdolib 0 - - Using software emulation of serial port instead of h/w UART
<134>1 - Garage-Door-F26E04 ratgdo-comms 0 - - id code 1127737 (0x113539)
<134>1 - Garage-Door-F26E04 ratgdo-comms 0 - - rolling code 7785 (0x1E69)
<134>1 - Garage-Door-F26E04 ratgdo-comms 0 - - Save rolling code: 0
<134>1 - Garage-Door-F26E04 sw_serial 0 - - Set serial port protocol to Unknown
<134>1 - Garage-Door-F26E04 sw_serial 0 - - Set serial port protocol to Security+ 2.0
<134>1 - Garage-Door-F26E04 sw_serial 0 - - Starting GDO software serial task.
<134>1 - Garage-Door-F26E04 gdolib 0 - - GDO Started
<134>1 - Garage-Door-F26E04 sw_serial 0 - - Set serial port protocol to Security+ 2.0
<134>1 - Garage-Door-F26E04 ratgdo-http 0 - - === Starting HTTP web server ===
<134>1 - Garage-Door-F26E04 ratgdo-http 0 - - Allocated buffer for JSON, size: 2048
<134>1 - Garage-Door-F26E04 ratgdo-http 0 - - Motion triggers, motion : 0, obstruction: 0, light key: 0, door key: 0, lock key: 0, asInt: 0
<134>1 - Garage-Door-F26E04 ratgdo-http 0 - - Registering URI handlers
<134>1 - Garage-Door-F26E04 ratgdo-http 0 - - Free heap (HTTP server started): 109972
<134>1 - Garage-Door-F26E04 ratgdo-homekit 0 - - Status: Need to pair
<134>1 - Garage-Door-F26E04 gdolib 0 - - SYNC TASK: Getting status
<134>1 - Garage-Door-F26E04 ratgdo-main 0 - - Free heap dropped to 109944
<134>1 - Garage-Door-F26E04 gdolib 0 - - SYNC TASK: Getting status
<134>1 - Garage-Door-F26E04 gdolib 0 - - SYNC TASK: Getting status
<134>1 - Garage-Door-F26E04 gdolib 0 - - SYNC TASK: Getting status
<134>1 - Garage-Door-F26E04 ratgdo-http 0 - - Client 10.1.1.30 requesting: /index.html (HTTP_GET, type: text/html, length: 6825)
<134>1 - Garage-Door-F26E04 gdolib 0 - - SYNC TASK: Getting status
<134>1 - Garage-Door-F26E04 ratgdo-http 0 - - Client 10.1.1.30 requesting: /status.json (method: HTTP_GET)
<134>1 - Garage-Door-F26E04 ratgdo-http 0 - - JSON length: 1300
<134>1 - Garage-Door-F26E04 ratgdo-http 0 - - Client 10.1.1.30 requesting: /rest/events/subscribe (method: HTTP_GET)
<134>1 - Garage-Door-F26E04 ratgdo-http 0 - - SSE Subscription for client 795924ea-ae59-45a4-88c5-b7cbec53a2f7 with IP 10.1.1.30: event bus location: /rest/events/0, Total subscribed: 1
<134>1 - Garage-Door-F26E04 ratgdo-http 0 - - Client 10.1.1.30 listening for SSE events on channel 0
<134>1 - Garage-Door-F26E04 ratgdo-main 0 - - Free heap dropped to 106516
<134>1 - Garage-Door-F26E04 gdolib 0 - - SYNC TASK: Getting status
<134>1 - Garage-Door-F26E04 gdolib 0 - - SYNC TASK: Getting status
<134>1 - Garage-Door-F26E04 gdolib 0 - - SYNC TASK: Getting status
<134>1 - Garage-Door-F26E04 gdolib 0 - - SYNC TASK: Getting status
<134>1 - Garage-Door-F26E04 gdolib 0 - - SYNC TASK: Getting status
<134>1 - Garage-Door-F26E04 ratgdo-comms 0 - - GDO event: synced: false, protocol: Security+ 2.0
<134>1 - Garage-Door-F26E04 ratgdo-comms 0 - - Client ID: 1127737, Rolling code: 7796
<134>1 - Garage-Door-F26E04 ratgdo-comms 0 - - Rolling code set to 7896, retrying sync
<134>1 - Garage-Door-F26E04 ratgdo-comms 0 - - Save rolling code: 7896
<134>1 - Garage-Door-F26E04 sw_serial 0 - - Set serial port protocol to Security+ 2.0
<134>1 - Garage-Door-F26E04 gdolib 0 - - SYNC TASK: Getting status
<134>1 - Garage-Door-F26E04 gdolib 0 - - SYNC TASK: Getting status
<134>1 - Garage-Door-F26E04 gdolib 0 - - SYNC TASK: Getting status
<134>1 - Garage-Door-F26E04 gdolib 0 - - SYNC TASK: Getting status
<134>1 - Garage-Door-F26E04 gdolib 0 - - SYNC TASK: Getting status
<134>1 - Garage-Door-F26E04 gdolib 0 - - SYNC TASK: Getting status
<134>1 - Garage-Door-F26E04 gdolib 0 - - SYNC TASK: Getting status
<134>1 - Garage-Door-F26E04 gdolib 0 - - SYNC TASK: Getting status
<134>1 - Garage-Door-F26E04 gdolib 0 - - SYNC TASK: Getting status
<134>1 - Garage-Door-F26E04 gdolib 0 - - SYNC TASK: Getting status
<134>1 - Garage-Door-F26E04 ratgdo-comms 0 - - GDO event: synced: false, protocol: Security+ 2.0
<134>1 - Garage-Door-F26E04 ratgdo-comms 0 - - Client ID: 1127737, Rolling code: 7907
[... 756 lines cut ...]
<134>1 - Garage-Door-F26E04 ratgdo-vehicle 0 - - Vehicle status: Parked
<134>1 - Garage-Door-F26E04 ratgdo-http 0 - - SSE send to client 10.1.1.30 on channel 0, data: { "vehicleStatus": "Parked", "upTime": 149896 }
<134>1 - Garage-Door-F26E04 ratgdo-http 0 - - SSE send to client 10.1.1.173 on channel 1, data: { "vehicleStatus": "Parked", "upTime": 149896 }
<134>1 - Garage-Door-F26E04 gdolib 0 - - cmd=080 (GET_STATUS) byte2=01 byte1=00 nibble=0
<134>1 - Garage-Door-F26E04 gdolib 0 - - cmd=392 (PING) byte2=00 byte1=00 nibble=0
[... cut ...]
<134>1 - Garage-Door-F26E04 gdolib 0 - - cmd=48c (OPENINGS) byte2=c4 byte1=1b nibble=0
<134>1 - Garage-Door-F26E04 gdolib 0 - - cmd=308 (PAIRED_DEVICES) byte2=0b byte1=00 nibble=0
<134>1 - Garage-Door-F26E04 ratgdo-vehicle 0 - - Vehicle status: Away
<134>1 - Garage-Door-F26E04 ratgdo-http 0 - - SSE send to client 10.1.1.30 on channel 0, data: { "vehicleStatus": "Away", "upTime": 188223 }
<134>1 - Garage-Door-F26E04 ratgdo-http 0 - - SSE send to client 10.1.1.173 on channel 1, data: { "vehicleStatus": "Away", "upTime": 188223 }

```

dkerr64, 2025-04-29:
Thank you for the log. If there is a way to configure your syslog server to insert timestamps please do that.  Not sure why every line starts with <134>1.

This log shows two boots (search on WiFi established... that is the very first message that will go to syslog server on a boot).  The first boot never completes.  One thing I notice is that it takes a very long time to sync with the door, lots of _SYNC TASK: Getting status_ messages.  I usually see a few of these during boot, but nothing like the number you see.  @descipher any thoughts on this?

There are also quite a lot of _collision detected_ messages.  Were you actually connecting wires with it powered on?  That may explain that, but hard to tell as there are no timestamps.

By the end of the log things look normal.

But is it stable long term?  Can you do a reboot and see that it comes back clean?

If you are able to get your syslog server to add timestamps, please do.  Then lets leave it running until problem occurs and then we will look at the log again.

ryanburnett, 2025-04-29:
I'm using Syslog Watcher Manager. I can copy the messages as JSON (attached above), CSV, tabbed values, or raw text. Happy to provide another file format instead of JSON.

Yes, I was connecting the wires with it powered on because I wanted to test the power first, then I connected the wires afterwards.

I just rebooted the disco but it's still not logging anything to the server.

GelidusResearch, 2025-04-29:
With the wireline in an unconnected state it would create a barrage of messages the likes of what was logged. Such as status retries and pairing fails, it would also put the sync out of joint requiring a reboot to regain secplus comms. The loss of a WiFi connection would be a separate issue and that one needs analysis.  To have some idea of whats going on requires log at the time of failure.  With a WiFi disconnect event it would be useful to know some of the network topology info and signal strength. Is it a mesh? etc.

ryanburnett, 2025-04-30:
Here's a screenshot of the web UI:

[https://imgur.com/a/XyqEvvv](url)

The signal strength is around -42 dBm. I have a:

- UniFi Dream Machine SE gateway
- Pro HD 24 PoE switch (connected to the SE with SFP+)
- U7 Pro XGS access point (in the garage, connected to the switch with Cat6A)

Please let me know if there's any additional info I can provide. Thanks.

dkerr64, 2025-04-30:
>The loss of a WiFi connection would be a separate issue and that one needs analysis. To have some idea of whats going on requires log at the time of failure. With a WiFi disconnect event it would be useful to know some of the network topology info and signal strength. Is it a mesh? etc.

I recently made some changes that should make the homekit-ratgdo32 more tolerant of WiFi outages.  I have UniFi AP's and they sometimes do a channel scan in the middle of the night, disconnecting clients.  I've not thoroughly tested it, but it used to repeat initial setup (gdo comms, vehicle distance sensor, etc.) on reconnect when it should not have.

On the WiFi established log message there is a count reported.  1 means first connection after reboot, any higher value will indicate that WiFi was disconnected and reconnected due to some external event, like I describe above.

dkerr64, 2025-04-30:
> Please let me know if there's any additional info I can provide. Thanks.

We need to let it run, logging to syslog, and if the problem occurs again then save the logs.

GelidusResearch, 2025-04-30:
-41dBm is nearly 100% signal. No know issues with the U7 Pro XGS.

ryanburnett, 2025-05-06:
I think I figured out the problem.

The disco has been running perfectly with the HomeKit firmware for over a week.

A few hours ago I used the web UI to rename the GDO. I renamed the GDO to "Ryan's Garage Door" and then it went offline about an hour later.

I also renamed it to "Ryan's Garage Door" a week ago, it went offline, and I made this ticket. I think the apostrophe (i.e. "'" with no quotes) caused the GDO to go offline.

I'll install the HomeKit firmware again, not rename the GDO, and that should fix it.

So I think renaming the GDO with an apostrophe is causing the GDO to go offline.

closed 2025-05-06 by ryanburnett

ryanburnett, 2025-05-06:
Thanks David. Both discos are working perfectly now. I really appreciate all of the hard work from you and the team.