ratgdo/homekit-ratgdo32 #81: LiftMaster LM80EVFA - EU version, door closing does not work

hejsiri, 2025-06-23:
I installed a ratgdo32 controller with HomeKit software to my LiftMaster LM80EVFA automatic, but I have a problem with the shutdown.

It works:
- turn the light on / off
- open door
- door statuses are correct

Does not work
- closing the door

I also have doubts about the wiring connection. The manual says that in the EU version the red wire should be connected to connector 0 (green) instead of connector 1 (red). However, there is no information on where I should connect the black wire. I connected it to connector 3 (gray), but it did not change the situation.

What could be the reason that the door closing does not work?

Below is a photo of my automatic.

[image]
[image]

dkerr64, 2025-06-28:
Looking at the log, I see both door opening and closing.... I assume that those logs are generated by you requesting door open/close from the wall panel or a remote.  But that requesting a close from the ratgdo32 web page or HomeKit is not working.  Correct?

As there is a lot of noise in the log, it would help if you got the door into a known state using the wall panel.... open it, then close it.  Now check the web page shows correct state and start log from that point... then open the door from the web page, identify in the log the timestamp you did that... and then close the door (or try to) from the web page, again note the timestamp when you did that.

Also, please set time-to-close to zero, just to eliminate that code path and simplify diagnosis.

Thanks

dkerr64, 2025-07-02:
I'm a s/w guy, not h/w, so cannot comment on whether the black wire needs to be connected or not.  @PaulWieland could advise.

However... logs show that the firmware is sending/receiving data.  So in this particular case it may not make a difference.

dkerr64, 2025-07-02:
@PaulWieland that code segments suggests that I can use "toggle" in the code instead of "close" and it will work?

hejsiri, 2025-07-03:
> ROW openers will not respond to the door close command if they don't have obstruction sensors installed. NA openers wont close no matter what without obstruction sensors.
> 
> See here: https://github.com/ratgdo/esphome-ratgdo/blob/64648e2d1754ee9e499204dfd9e87800427d8243/components/ratgdo/ratgdo.cpp#L556

Can you explain the abbreviations ROW and NA? :)

> obstruction sensors 

Do you mean photo eyes like the ones in this picture?

[image]

PaulWieland, 2025-07-03:
> [@PaulWieland](https://github.com/PaulWieland) that code segments suggests that I can use "toggle" in the code instead of "close" and it will work?

Yes. If no obstruction sensors are detected, use the toggle command instead of close.

hejsiri, 2025-07-03:
> @dkerr64: And the photograph of the wiring to your garage door opener (GDO) suggests that you do not have any obstruction sensors connected.

obstruction sensor is necessary? My drive is not equipped with an obstruction sensor. Should I buy it?

> @PaulWieland: Yes. If no obstruction sensors are detected, use the toggle command instead of close.

If I understand correctly, closing should work without a sensor.

hejsiri, 2025-07-04:
I have a LiftMaster LM80EVFA. Being at a friend's house, I noticed completely by chance that he has the same Liftmaster model in the garage, what's more he has the usual wired wall buttons connected to it. It turns out that my opener does not require ratgdo with Security+ 2.0 protocol , and can work with ordinary Shelly 2PM + 2 sensors and HomeKit soft. In ratgdo HK software I switched to dry contact and it works, but it requires connecting additional wired sensors, so it's pointless because it doesn't see the states correctly I wanted to avoid putting on additional open/close sensors.

I bought an expensive toy that doesn't work :(

dkerr64, 2025-07-04:
@hejsiri please try release 3.2.0.  This is currently marked as "pre-release" so you will need to click on the update firmware button, then check "include pre-releases", press check for update button, then you can do the update.

With this release I have added a user setting, under the "door protocol" section select the Use TOGGLE checkbox and save.  Now try operating your door.  A reboot is not necessary on changing this setting.

I have made this user selectable for now, rather than automatically detecting whether obstruction sensor is present like the ESPHome code... it was just easier for now.

Thanks

hejsiri, 2025-07-05:
I did the upgrade to 3.2.0 and now it works perfectly. Thank you very much, great job :)