ratgdo/homekit-ratgdo32 #93: Feature Request: mDNS / Bonjour hostname advertisement

nickv2002, 2025-09-21:
It would be great if ratgdo/homekit-ratgdo32 consistently advertised its web interface hostname via mDNS (e.g. garage-door-ABCDEF.local) and published an HTTP service record (_http._tcp.local) so that users can access the device web interface using a stable .local name, instead of having to track its IP address.

Use-case:
	•	The firmware already allows setting a Name in the web UI which becomes the “network hostname” (truncated, sanitized) per README. But in practice, it appears .local hostname may not be being advertised (or discoverable).
	•	When DHCP assigns a different IP, users have to scan or inspect router settings to find the device. A .local name would simplify that a lot.

Proposed implementation:
	•	On device WiFi connect, call MDNS.begin(<device_name>), using the same name as the “Name” setting.
	•	Add MDNS.addService("http", "tcp", 80).
	•	Ensure the device_name sanitization matches existing logic (max length, non-alphanumeric replaced by hyphens).
	•	Ensure persistence across reboots.
	•	Provide option to disable mDNS if needed for resource constrained cases.

Possible challenges / considerations:
	•	Memory / code size constraints on ESP32: mDNS libraries may consume RAM; ensure minimal footprint.
	•	Interaction with existing HomeKit mDNS/HAP advertising (avoid namespace / port conflicts).
	•	Networks that block multicast or have isolated client isolation could inhibit .local resolution. Note in docs that the feature depends on network supporting mDNS / multicast DNS

dkerr64, 2025-09-22:
This turns out to be a one-line add... at least on the ESP32-based firmware.  Not sure about the original ESP8266.

nickv2002, 2025-09-29:
Awesome, works well though I would add two very minor suggestions (which I can create another issue for but figured I'd start here)

1. The UI doesn't seem to _tell_ the user that they can use the .local hostname they just have to know to try it
2. If you change your hostname in the webUI Settings, it doesn't seem to update the mDNS listing, I had to reboot my device for the new mDNS hostname to get picked up.

But these are minor. Thanks for the support and quick turnaround!

dkerr64, 2025-09-29:
> Awesome, works well though I would add two very minor suggestions (which I can create another issue for but figured I'd start here)
>
> 1. The UI doesn't seem to _tell_ the user that they can use the [...]

Yes, good catch.  I will look at updating it without requiring a reboot.

Not sure where it should be noted on the UI though?  Any suggestions?

nickv2002, 2025-10-06:
> > 1. The UI doesn't seem to _tell_ the user that they can use the .local hostname they just have to know to try it

> Not sure where it should be noted on the UI though? Any suggestions?

Maybe just a link in the WebUI is enough? [I had an LLM draft up an untested patch if that helps](https://github.com/ratgdo/homekit-ratgdo32/pull/104).