ratgdo/homekit-ratgdo32 #95: Opening Status Not Reported Correctly in Homekit.

jasonbyun8, 2025-09-24:
This issue should have been fixed by https://github.com/ratgdo/homekit-ratgdo32/issues/53, but it does not seem to work for me;
1. Door state still unknown

[image]

2. Intermittent lock/unlock notification 
[image]

3. Close/Open door command just does not work sometimes, will have to call this twice to make it working.

Some log outputs

1. _____ _____ _____ _____ ____  _____
| __  |  _  |_   _|   __|    \|     |
|    -|     | | | |  |  |  |  |  |  |
|__|__|__|__| |_| |_____|____/|_____|
           for HomeKit
Garage Door AD100C

System Log
Status
Reboot Log
Crash Log
 
Current server message log...
Server uptime: 125425 ms (00:02:05.425)
Firmware version: 3.3.3
Free heap: 100196
Minimum heap: 99068
1:06.276) ratgdo-comms: SEC1 RX invalid cmd byte 0x53
D (00:01:06.286) ratgdo-comms: SEC1 RX invalid cmd byte 0x20
D (00:01:07.069) ratgdo-comms: SEC1 RX invalid cmd byte 0x06
D (00:01:07.319) ratgdo-comms: SEC1 RX invalid cmd byte 0x40
D (00:01:07.329) ratgdo-comms: SEC1 RX invalid cmd byte 0x00
D (00:01:07.581) ratgdo-comms: SEC1 RX invalid cmd byte 0x53
D (00:01:07.591) ratgdo-comms: SEC1 RX invalid cmd byte 0x20
D (00:01:08.374) ratgdo-comms: SEC1 RX invalid cmd byte 0x06
D (00:01:08.624) ratgdo-comms: SEC1 RX invalid cmd byte 0x40
D (00:01:08.637) ratgdo-comms: SEC1 RX invalid cmd byte 0x00
D (00:01:08.886) ratgdo-comms: SEC1 RX invalid cmd byte 0x53
D (00:01:08.896) ratgdo-comms: SEC1 RX invalid cmd byte 0x20
D (00:01:09.682) ratgdo-comms: SEC1 RX invalid cmd byte 0x06
D (00:01:09.932) ratgdo-comms: SEC1 RX invalid cmd byte 0x40
D (00:01:09.942) ratgdo-comms: SEC1 RX invalid cmd byte 0x00
D (00:01:10.194) ratgdo-comms: SEC1 RX invalid cmd byte 0x53
D (00:01:10.205) ratgdo-comms: SEC1 RX invalid cmd byte 0x20
D (00:01:10.987) ratgdo-comms: SEC1 RX invalid cmd byte 0x06
D (00:01:11.240) ratgdo-comms: SEC1 RX invalid cmd byte 0x40
D (00:01:11.250) ratgdo-comms: SEC1 RX invalid cmd byte 0x00
D (00:01:11.502) ratgdo-comms: SEC1 RX invalid cmd byte 0x53
D (00:01:11.510) ratgdo-comms: SEC1 RX invalid cmd byte 0x20
D (00:01:12.298) ratgdo-comms: SEC1 RX invalid cmd byte 0x06
D (00:01:12.545) ratgdo-comms: SEC1 RX invalid cmd byte 0x40
D (00:01:12.555) ratgdo-comms: SEC1 RX invalid cmd byte 0x00
D (00:01:12.807) ratgdo-comms: SEC1 RX invalid cmd byte 0x53
D (00:01:12.818) ratgdo-comms: SEC1 RX invalid cmd byte 0x20
D (00:01:13.603) ratgdo-comms: SEC1 RX invalid cmd byte 0x06
D (00:01:13.853) ratgdo-comms: SEC1 RX invalid cmd byte 0x40
D (00:01:13.863) ratgdo-comms: SEC1 RX invalid cmd byte 0x00
D (00:01:14.112) ratgdo-comms: SEC1 RX invalid cmd byte 0x53
D (00:01:14.123) ratgdo-comms: SEC1 RX invalid cmd byte 0x20
D (00:01:14.905) ratgdo-comms: SEC1 RX invalid cmd byte 0x06
D (00:01:15.158) ratgdo-comms: SEC1 RX invalid cmd byte 0x40
D (00:01:15.168) ratgdo-comms: SEC1 RX invalid cmd byte 0x00
D (00:01:15.420) ratgdo-comms: SEC1 RX invalid cmd byte 0x53
D (00:01:15.428) ratgdo-comms: SEC1 RX invalid cmd byte 0x20
D (00:01:16.213) ratgdo-comms: SEC1 RX invalid cmd byte 0x06
D (00:01:16.463) ratgdo-comms: SEC1 RX invalid cmd byte 0x40
D (00:01:16.473) ratgdo-comms: SEC1 RX invalid cmd byte 0x00
D (00:01:16.722) ratgdo-comms: SEC1 RX invalid cmd byte 0x53
[... 640 lines cut ...]
D (00:38:16.124) ratgdo-comms: SEC1 RX invalid cmd byte 0x40
D (00:38:16.136) ratgdo-comms: SEC1 RX invalid cmd byte 0x00
D (00:38:16.385) ratgdo-comms: SEC1 RX invalid cmd byte 0x53
D (00:38:16.397) ratgdo-comms: SEC1 RX invalid cmd byte 0x20
D (00:38:17.178) ratgdo-comms: SEC1 RX invalid cmd byte 0x08
D (00:38:17.431) ratgdo-comms: SEC1 RX invalid cmd byte 0x40
D (00:38:17.443) ratgdo-comms: SEC1 RX invalid cmd byte 0x00
D (00:38:17.690) ratgdo-comms: SEC1 RX invalid cmd byte 0x53
D (00:38:17.702) ratgdo-comms: SEC1 RX invalid cmd byte 0x20
D (00:38:18.484) ratgdo-comms: SEC1 RX invalid cmd byte 0x08
D (00:38:18.736) ratgdo-comms: SEC1 RX invalid cmd byte 0x40
D (00:38:18.748) ratgdo-comms: SEC1 RX invalid cmd byte 0x00
I (00:38:18.764) ratgdo-main: Free heap dropped to 87904
D (00:38:18.998) ratgdo-comms: SEC1 RX invalid cmd byte 0x53
D (00:38:19.010) ratgdo-comms: SEC1 RX invalid cmd byte 0x20
D (00:38:19.790) ratgdo-comms: SEC1 RX invalid cmd byte 0x08
D (00:38:20.043) ratgdo-comms: SEC1 RX invalid cmd byte 0x40
D (00:38:20.055) ratgdo-comms: SEC1 RX invalid cmd byte 0x00
D (00:38:20.304) ratgdo-comms: SEC1 RX invalid cmd byte 0x53
D (00:38:20.317) ratgdo-comms: SEC1 RX invalid cmd byte 0x20
D (00:38:21.099) ratgdo-comms: SEC1 RX invalid cmd byte 0x08
D (00:38:21.351) ratgdo-comms: SEC1 RX invalid cmd byte 0x40
D (00:38:21.363) ratgdo-comms: SEC1 RX invalid cmd byte

I could help further if there is more specific action you want me to perform for debugging; hope we can fix this problem sooner rather than later..

jasonbyun8, 2025-09-25:
> All of the invalid cmd byte messages indicate a probem receiving data from
> the garage door. @mitchjs <https://github.com/mitchjs> any suggestion on
> how Jason could clean up the signal he gets? There must be a lot of
> interference.
>
> @jasonbyun8 <https://github.com/jasonbyun8> it looks like you have a
> digital wall panel connected. For testing purposes, could you disconnect
> that and then reboot ratgdo... on the screen shot you sent the "Door
> Protocol: Sec+" line should then show the word "Emulated" that would signal
> that there is no digital wall panel. Then check the logs and see if you
> still get all these invalid cmd byte messages or not.
>
> Thanks

Disconnecting wall control seemed to resolve the problem:

Current server message log...

Server uptime: 104989 ms (00:01:44.989)
Firmware version: 3.3.3
Free heap: 94192
Minimum heap: 93172
141), Free Entries: (489), Total Entries: (630), Namespace Count: (9)
I (00:00:00.018) ratgdo-config: Read user configuration from NVRAM
I (00:00:00.034) ratgdo-utils:    deviceName:          Garage Door AD100C
I (00:00:00.034) ratgdo-utils:    wifiChanged:         false
I (00:00:00.044) ratgdo-utils:    wifiPower:           20
I (00:00:00.045) ratgdo-utils:    wifiPhyMode:         0
I (00:00:00.045) ratgdo-utils:    staticIP:            false
I (00:00:00.055) ratgdo-utils:    localIP:             192.168.1.35
I (00:00:00.066) ratgdo-utils:    subnetMask:          255.255.255.0
I (00:00:00.066) ratgdo-utils:    gatewayIP:           192.168.1.1
I (00:00:00.076) ratgdo-utils:    nameserverIP:        192.168.1.1
I (00:00:00.076) ratgdo-utils:    wwwPWrequired:       false
I (00:00:00.087) ratgdo-utils:    wwwUsername:         admin
I (00:00:00.087) ratgdo-utils:    wwwCredentials:
10d3c00fa1e09696601ef113b99f8a87
I (00:00:00.097) ratgdo-utils:    GDOSecurityType:     1
I (00:00:00.097) ratgdo-utils:    TTCseconds:          5
I (00:00:00.108) ratgdo-utils:    rebootSeconds:       3600
I (00:00:00.108) ratgdo-utils:    LEDidle:             0
I (00:00:00.118) ratgdo-utils:    motionTriggers:      2
I (00:00:00.118) ratgdo-utils:    obstFromStatus:      false
I (00:00:00.129) ratgdo-utils:    enableNTP:           false
I (00:00:00.129) ratgdo-utils:    doorUpdateAt:        0 ()
I (00:00:00.139) ratgdo-utils:    doorOpenAt:          0 ()
I (00:00:00.139) ratgdo-utils:    doorCloseAt:         0 ()
I (00:00:00.150) ratgdo-utils:    timeZone:
America/Los_Angeles;PST8PDT,M3.2.0,M11.1.0
I (00:00:00.160) ratgdo-utils:    softAPmode:          false
I (00:00:00.160) ratgdo-utils:    syslogEn:            false
I (00:00:00.171) ratgdo-utils:    syslogIP:            0.0.0.0
I (00:00:00.171) ratgdo-utils:    syslogPort:          514
I (00:00:00.181) ratgdo-utils:    vehicleThreshold:    100
I (00:00:00.182) ratgdo-utils:    vehicleHomeKit:      false
I (00:00:00.192) ratgdo-utils:    laserEnabled:        false
I (00:00:00.192) ratgdo-utils:    laserHomeKit:        true
I (00:00:00.202) ratgdo-utils:    assistDuration:      60
I (00:00:00.203) ratgdo-utils:    occupancyDuration:   0
I (00:00:00.203) ratgdo-utils:    enableIPv6:          false
I (00:00:00.213) ratgdo-utils:    homespanCLI:         false
I (00:00:00.213) ratgdo-utils: RFC952 device hostname: Garage-Door-AD100C
I (00:00:00.224) ratgdo-improv: Enable Improv for WiFi provisioning.
Disable HomeSpan serial logging and CLI
I (00:00:01.234) ratgdo-homekit: === Setup HomeKit accessories and services ===
I (00:00:01.235) ratgdo-homekit: HomeKit pairing QR Code ID: laWq
I (00:00:01.245) ratgdo-homekit: HomeKit QR setup payload: X-HM://0042WZMX3laWq
I (00:00:03.349) ratgdo-homekit: Configuring HomeKit Garage Door Service
I (00:00:03.350) ratgdo-homekit: Configuring HomeKit Light Service for GDO Light
I (00:00:03.361) ratgdo-homekit: Configuring HomeKit Motion Service for Motion
I (00:00:03.362) ratgdo-homekit: No vehicle presence sensor. Skipping
motion and occupancy services
I (00:00:03.376) ratgdo-main: Free heap dropped to 172880
I (00:00:07.245) ratgdo-homekit: Status: WiFi connecting
I (00:00:07.246) ratgdo-homekit: Wifi begin for SSID: ORBI92
I (00:00:07.347) ratgdo-homekit: WiFi TX power: 78
I (00:00:08.241) ratgdo-main: Free heap dropped to 121208
I (00:00:09.241) ratgdo-main: Free heap dropped to 121200
I (00:00:10.243) ratgdo-main: Free heap dropped to 120936
I (00:00:11.245) ratgdo-main: Free heap dropped to 120068
I (00:00:12.249) ratgdo-homekit: Wifi begin for SSID: ORBI92
I (00:00:12.263) ratgdo-homekit: WiFi TX power: 78
I (00:00:15.249) ratgdo-main: Free heap dropped to 120060
I (00:00:16.447) ratgdo-homekit: Received IPv4 Address: 192.168.1.35
[... 209 lines cut ...]
I (00:01:48.503) ratgdo-http: Client 192.168.1.36 requesting:
/logs.html (HTTP_GET, type: text/html, length: 1697)
I (00:01:48.604) ratgdo-http: Client 192.168.1.36 requesting:
/rest/events/subscribe (method: HTTP_GET)
I (00:01:48.616) ratgdo-http: Client 192.168.1.36
(48774634-b005-461c-a401-7b46a6d891f4) SSE subscription:
/rest/events/1, Total: 2, Heartbeat: 0, Log: 1
I (00:01:48.710) ratgdo-http: Client 192.168.1.36
(48774634-b005-461c-a401-7b46a6d891f4) listening for SSE events on
channel 1
I (00:01:48.805) ratgdo-http: Client 192.168.1.36 requesting: /showlog
(method: HTTP_GET)
I (00:01:48.931) ratgdo-http: Client 192.168.1.36 requesting:
/status.json (method: HTTP_GET)
I (00:01:48.955) ratgdo-http: JSON length: 1514, build time 4ms,
response time: 10ms
I (00:01:49.030) ratgdo-http: Client 192.168.1.36 requesting:
/showrebootlog (method: HTTP_GET)
I (00:01:49.131) ratgdo-http: Client 192.168.1.36 requesting:
/crashlog (method: HTTP_GET)
I (00:01:49.422) ratgdo-main: Free heap dropped to 92264
I (00:01:50.424) ratgdo-main: Free heap dropped to 90628

I am wanting to keep that capability though, and I thought it should work
with the wall remote. Any suggestion about this?

dkerr64, 2025-09-25:
@mitchjs if you look at the sequence it is very consistent...

```
D (00:02:06.338) ratgdo-comms: SEC1 RX invalid cmd byte 0x53
D (00:02:06.348) ratgdo-comms: SEC1 RX invalid cmd byte 0x20
D (00:02:07.134) ratgdo-comms: SEC1 RX invalid cmd byte 0x08
D (00:02:07.383) ratgdo-comms: SEC1 RX invalid cmd byte 0x40
D (00:02:07.393) ratgdo-comms: SEC1 RX invalid cmd byte 0x00
```

and those five bytes keep repeating.  Possibly we have a sequence we have not seen before?

jasonbyun8, 2025-09-25:
Not sure about the model number..

dkerr64, 2025-09-25:
[firmware.bin.zip](https://github.com/user-attachments/files/22527022/firmware.bin.zip)

Jason, attached is a firmware build that uses the external GDOLIB to communicate with the garage door... an alternative to our own internal comms code.  Could you try running with this please (with your wall panel connected), capturing the logs, and let us know if it works.  Again this is to help narrow down what may be going on and whether an alternative comms library works.

The attached file is zipped... you will need to unzip to firmware.bin then upload that.

Thanks.

dkerr64, 2025-09-25:
@mitchjs looking at the [ESPhome](https://github.com/ratgdo/esphome-ratgdo/blob/cd712c73ce93dee4befed7dc467421711d192f28/components/ratgdo/secplus1.cpp#L358) firmware, they are handling 0x37 commands that we are not... and from that are setting a flag `is_0x37_panel` which is then tested in several places.  Again something we are not doing.

Do you have time to take a look?  I'm wondering if Jason has one of these "0x37 panels" and we're missing support for it.

mitchjs, 2025-09-25:
gotta start with what exact wall panel you got
for sure your communications are a mess
is this a RATGDO32 or older RATGDO (ESP8266)?
prob will need more detailed debug on comm messages (debug level to debug at a minimum)

dkerr64, 2025-09-25:
@PaulWieland can you explain what the "0x37 panel" code is doing in ESPhome firmware?

jasonbyun8, 2025-09-25:
> I will do it after work, but when I checked yesterday there was no info on
> the back..
> … <#m_2528762416398059297_>
>
> thats even more odd, than the issues :)

Maybe I should’ve opened the back cover. let me get back to you

dkerr64, 2025-09-25:
More data points...

https://github.com/Kaldek/rat-ratgdo/discussions/95 (@mitchjs you in this one)

https://community.home-assistant.io/t/ratgdo-local-mqtt-control-for-your-chamberlain-liftmaster-security-2-0-garage-door-opener/437380/566?page=28

[image]

dkerr64, 2025-09-25:
And lets move further discussion / comment over to here please...
https://github.com/dkerr64/homekit-ratgdo32/pull/7

dkerr64, 2025-10-02:
I am closing this issue.  The next release will "be aware" of 0x37 door panels and so not misinterpret data received.  It also implements the same logic as the ESPhome firmware that attempts to get door status.  However, testing suggests that this is not a reliable solution.

Recommended fix is to replace the wall panel with a [LiftMaster 889LM](https://www.liftmaster.com/myq-control-panel/p/889LMMC) provided it is compatible with your door.  See attached PDF for compatibility list. 

[LM_Accessory_Compatibility_Chart_EN.pdf](https://github.com/user-attachments/files/22663781/LM_Accessory_Compatibility_Chart_EN.pdf)